Lora-net/LoRaMac-node #285: Correct way of using UART Drivers.

RakshitAdmar, 2017-07-30:
I've been struggling for a while getting the UART to properly work. 
At the moment, I have TX and RX working perfectly well independently, but the moment I try to echo back what ever I RX,  the the first two characters are always missing.
When my code only does RX, I receive all the characters I type correctly. But when I do a TX of what i receive after RX, only the first time all the characters are echoed back. After the first time, the first two characters are always missed. I've debugged the code, and found that the 

My UART Callback functions 

```
void HAL_UART_TxCpltCallback( UART_HandleTypeDef *handle )
{
    if( IsFifoEmpty( &Uart1.FifoTx ) == false )
    {
        TxData = FifoPop( &Uart1.FifoTx );
        //  Write one byte to the transmit data register
        HAL_UART_Transmit_IT( &UartHandle, &TxData, 1 );
    }

    if(TxData == 10)
        {
        	if( Uart1.IrqNotify != NULL )
        		Uart1.IrqNotify( UART_NOTIFY_TX );
        }
}

void HAL_UART_RxCpltCallback( UART_HandleTypeDef *handle )
{
    if( IsFifoFull( &Uart1.FifoRx ) == false )
    {
        // Read one byte from the receive data register
        FifoPush( &Uart1.FifoRx, RxData );
    }

    if(RxData == 10)
    {
    	if( Uart1.IrqNotify != NULL )
    		Uart1.IrqNotify( UART_NOTIFY_RX );
    }

    HAL_UART_Receive_IT( &UartHandle, &RxData, 1 );
}

```
My UART IRQNotify

```
void UartISR(UartNotifyId_t id){

	if(id == UART_NOTIFY_RX){
		flag =1;

}
	if(id == UART_NOTIFY_TX){
		flag=0;

}

}

```

How i do uart in main()

```
  FifoInit( &Uart1.FifoTx, txBuffer, 100);
  FifoInit( &Uart1.FifoRx, rxBuffer, 100);
  UartInit( &Uart1, 0, UART_TX, UART_RX );
  Uart1.IrqNotify = UartISR;
  UartConfig( &Uart1, RX_TX, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );

while(1)
{
if(flag == 1){

    		UartGetBuffer( &Uart1, rec_msg, 10, &msgLen );
    		UartPutBuffer( &Uart1, rec_msg,  msgLen);
    		flag = 0;
    	}
}

```

I've set the power supply to be USB_POWER in board.c

RakshitAdmar, 2017-07-30:
I'm sorry i asked this in haste. 
I seem to have found a fix. 
After UartPutBuffer is executed, there is no piece of code to re-enable HAL_UART_Receive_IT. 
I don't know why but it is somehow re-enabled after receiving 2 characters on the next cycle. 

In file uart-board.c, i implemented a simple function to re-enable uart rx

```
void UartReEnableRx(){
	HAL_UART_Receive_IT( &UartHandle, &RxData, 1 );

}

```

and execute it in my IRQ Notification handler, ref. above
```
void UartISR(UartNotifyId_t id){

	if(id == UART_NOTIFY_RX){
		flag =1;
	}
	
	if(id == UART_NOTIFY_TX){
		flag=0;
		UartReEnableRx();
	}

}

```

I strongly suggest this to be implemented in future releases, since otherwise, UART implementation of this driver will still remain to be in blocking mode.

mluis1, 2017-07-31:
If you take a closer look you will notice that the UART driver Put/Get functions return status values.
Which means that they must be checked in order to verify if the UART handling is ready or not to process a new command.

Please find below an example on how to make an UART echo application using this driver. 
Please note that we have used something similar to it in order to verify that the driver was implemented correctly.

```c
#include "board.h"

/*!
 * UART object
 */
Uart_t Uart1;

/*!
 * UART FIFO buffers size
 */
#define UART1_FIFO_TX_SIZE                                8
#define UART1_FIFO_RX_SIZE                                256

uint8_t Uart1TxBuffer[UART1_FIFO_TX_SIZE];
uint8_t Uart1RxBuffer[UART1_FIFO_RX_SIZE];

int main( void )
{
    BoardInitMcu( );
    BoardInitPeriph( );

    FifoInit( &Uart1.FifoTx, Uart1TxBuffer, UART1_FIFO_TX_SIZE );
    FifoInit( &Uart1.FifoRx, Uart1RxBuffer, UART1_FIFO_RX_SIZE );
    // Configure your terminal for 8 Bits data (7 data bit + 1 parity bit), no parity and no flow ctrl
    UartInit( &Uart1, UART_1, UART_TX, UART_RX );
    UartConfig( &Uart1, RX_TX, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );

    while( 1 )
    {
        uint8_t d = 0;
        if( UartGetChar( &Uart1, &d ) == 0 )
        {
            while( UartPutChar( &Uart1, d ) != 0 ){ };
        }
    }
    return 0;
}
```

Please also note that the Receive interruption is enabled at initialization of the driver and every time the Rx interrupt is handled by the HAL_UART_RxCpltCallback( UART_HandleTypeDef *handle ) function.

There is no need to change the drivers.

RakshitAdmar, 2017-07-31:
Thank you for showing me the correct way of performing UART. 
I did also try what you have just shown. As i mentioned earlier, the status was returning 0(OK) but two characters of the message i typed on my terminal were always missing. 
It is possible that it is because i wanted an interrupt to signal me when a complete line terminated with \n was received, and i was doing something wrong with my implementations. 
My only argument is that in file uart-board.c:204, `HAL_UART_RECEIVE_IT()` was not called once I TX'd so I missed a few leading characters. I'm not sure why it happened. 
I will confirm tomorrow and will leave this issue open until then.

Regardless, this is a fantastic driver implementation. Possibly because it is new, there isn't enough example applications for the controller peripherals. Thanks again for clearing this up.

closed 2017-08-04 by mluis1