Lora-net/LoRaMac-node #286: RegionNextChannel / RegionCommonUpdateBandTimeOff and nextTxDelay

jamesl-dm, 2017-07-31:
Hi,

I'm not 100% sure about this, because the timeout code is confusing. But I think there are two remaining issues:

1. RegionCommonUpdateBandTimeOff appears to be designed to reset the TimeOff counters of all bands that have completed their timeouts, and then return the minimum timeout from all of the bands.
However, the local variable nextTxDelay which holds the minimum timeout from all the bands is not assigned in all cases, so it can sometimes return infinity (-1) despite bands being available for immediate use.

2. The various RegionNextChannel implementations use CountNbOfEnabledChannels to find how many channels are available for immediate use, and how many would be available for use after waiting for their timeouts, **at a particular datarate**. If no channels are found for immediate use, but some will be available after a delay, the sleep time is updated to nextTxDelay. However, nextTxDelay was the minimum sleep time for **all channels at all datarates**. So the sleep time might be too short.

djaeckle, 2017-08-04:
Hi jamesl-dm,

thanks for your report. I agree that this functionality is not implemented in an easy an clear way. However, our motivation is to upate the implementation with an easier approach in the future. Regarding your questions:

1. As you already mentioned, the function `RegionCommonUpdateBandTimeOff` updates the time off counters of all bands. It's correct, that the variable `nextTxDelay` might not be updated in the function. This could happen if the time off values of all bands are elapsed. Assuming this case and assuming the current channel settings are supported, the end node will find a suitable channel e.g. in function RegionEU868NextChannel`. Then, `nextTxDelay` will not be handled because `nbEnabledChannels ` > 0 - so the LoRaMAC doesn't have to delay the transmission. 
So basically, the functionalty for the back off is spread around some functions. In the future, we want to remove such dependencies in order to have a clean implementation. Proposals to improve this functionality are welcome.

2. > The various RegionNextChannel implementations use CountNbOfEnabledChannels to find how many channels are available for immediate use, and how many would be available for use after waiting for their timeouts, at a particular datarate. If no channels are found for immediate use, but some will be available after a delay, the sleep time is updated to nextTxDelay. 

I agree

> However, nextTxDelay was the minimum sleep time for all channels at all datarates. So the sleep time might be too short.

I would describe a little bit different: function `RegionCommonUpdateBandTimeOff ` calculates the minimum time to wait before the LoRaMAC node is allowed to perform the next uplink. It takes the time off values of all bands into account. The value of `nextTxDelay` depends on uplink transmissions in the past. This delay is independent from the upcoming transmission - we need to wait at least the minimum time. After this delay, assuming the time off value of a band supporting the current datarate is zero, the node is allowed to perform the uplink. In contrast, it could happen that the time off value of a band is zero, but the band does not support the current datarate. Then the LoRaMAC shall delay the transmission again.
In summary, I think applying the minimum time is correct, even if it too short in some cases.

jamesl-dm, 2017-08-07:
Ah - Thanks for the explanation. I didn't realize that 1. and 2. worked together in that way. I thought that if the band didn't support the current data rate, there would be one short sleep, and then a tight loop. But since RegionCommonUpdateBandTimeOff doesn't set nextTxDelay to zero, there will be no tight loop.

So the only loose end is the error case when no data rate is supported, and RegionXxxNextChannel should return false:

```
bool RegionAU915NextChannel( NextChanParams_t* nextChanParams, uint8_t* channel, TimerTime_t* time, TimerTime_t* aggregatedTimeOff )
{
    ...
    TimerTime_t nextTxDelay = 0;
    ...   

    if( nextChanParams->AggrTimeOff <= TimerGetElapsedTime( nextChanParams->LastAggrTx ) )
    {
        ....
        // Update bands Time OFF
        nextTxDelay = RegionCommonUpdateBandTimeOff( nextChanParams->Joined, nextChanParams->DutyCycleEnabled, Bands, AU915_MAX_NB_BANDS );

        // Search how many channels are enabled
        nbEnabledChannels = CountNbOfEnabledChannels( nextChanParams->Datarate,
                                                      ChannelsMaskRemaining, Channels,
                                                      Bands, enabledChannels, &delayTx );
    }
    else
    {
        ....
    }

    if( nbEnabledChannels > 0 )
    {
        ....
        return true;
    }
    else
    {
        if( delayTx > 0 )
        {
            // Delay transmission due to AggregatedTimeOff or to a band time off
            *time = nextTxDelay;
            return true;
        }
        // Datarate not supported by any channel
        *time = 0;
        return false;
    }
}
```

I suggest that `if( delayTx > 0 )` should be `if( ( delayTx > 0 ) && ( delayTx != (uint32_t)(-1) ) )`.
Otherwise, if the data rate is not supported by any channel, and all bands are available for immediate use, *time becomes infinity?

djaeckle, 2017-08-07:
Hi jamesl-dm,

thanks for your feedback, Please note that `delayTx ` is only a counter. Function `CountNbOfEnabledChannels` computes the value for it and it contains the number of channels which must be delayed.

If the datarate is not supported by any channel, the value for `delayTx  `and `nbEnabledChannels ` will be zero. Then we don't care about the value of `nextTxDelay`, since the function `RegionAU915NextChannel `will return zero. Please note that in this case there may be bands available for immediate use.

The condition `if( delayTx > 0 )` can only be ture, if there is a band time off and if the current datarate is not available for immediate use. If the datarate is supported by the band which contains a time off, the variable `nextTxDelay` contains a valid value, computed by the function `RegionCommonUpdateBandTimeOff`.

Again, I want to emphasize that our motivation is to resolve mixed-up code in the future. Furthermore, we want to relocate code parts which are common to all regions into common implementations.

jamesl-dm, 2017-08-07:
Looks like I got my variable names confused....
Thanks so much for taking the time to explain this.

closed 2017-08-07 by jamesl-dm