Lora-net/LoRaMac-node #287: Missing Fcnt while testing with multiple devices

vikasrohal, 2017-07-31:
Hi,
I am testing with 5 End Devices and observation is : 
Some times Fcnt missed the sequence.
Suppose Send frame with Fcnt=10 and next frame will be with Fcnt=12 hence Fcnt=11 droped.  

Lora Gateway(Hardware) = iC880A with raspberry Pi3
Lora Packet Forwarder version v4.0.1
Lora Gateway version v5.0.1 

As per Attached Log 
Fcnt=3 send after Fcnt=1 , Fcnt=2 droped.
 
[image]

mluis1, 2017-07-31:
Could you please indicate which branch of this repository are you using?

Last week we solved a similar issue on the develop branch. We don't know if the issue you are observing is the same but, just in case please find below the commit message and fix that we will push on GitHub as soon as we have fully validated it.

> AckTimeoutRetriesCounter  must be reset every time a new request (unconfirmed or confirmed) is performed.
>
> If the counter isn't reset when switching between Confirmed and Unconfirmed messages we could end-up by increment the UpLinkCounter by 2.
> This situation happens only if one of the PrepareRxDoneAbort conditions is fulfilled between Confirmed messages and Unconfirmed messages.

The fix is the following:
```c
LoRaMacStatus_t LoRaMacMcpsRequest( McpsReq_t *mcpsRequest )
{
...
//************************Fix begin***************************
    // AckTimeoutRetriesCounter must be reset every time a new request (unconfirmed or confirmed) is performed.
    AckTimeoutRetriesCounter = 1;
//************************Fix end ***************************
    switch( mcpsRequest->Type )
    {
        case MCPS_UNCONFIRMED:
        {
            readyToSend = true;
            AckTimeoutRetries = 1;

            macHdr.Bits.MType = FRAME_TYPE_DATA_UNCONFIRMED_UP;
            fPort = mcpsRequest->Req.Unconfirmed.fPort;
            fBuffer = mcpsRequest->Req.Unconfirmed.fBuffer;
            fBufferSize = mcpsRequest->Req.Unconfirmed.fBufferSize;
            datarate = mcpsRequest->Req.Unconfirmed.Datarate;
            break;
        }
        case MCPS_CONFIRMED:
        {
            readyToSend = true;
//************************Fix begin***************************
            //AckTimeoutRetriesCounter = 1;
//************************Fix end***************************
            AckTimeoutRetries = mcpsRequest->Req.Confirmed.NbTrials;

            macHdr.Bits.MType = FRAME_TYPE_DATA_CONFIRMED_UP;

...
}
```

vikasrohal, 2017-08-01:
I am using latest version v4.3.2 released in April

mluis1, 2017-08-02:
Thanks for the info.

Could you please try the proposed fix by modifying your source code?
Please let us know if it solved your issue.

As you may be aware of the 4.3.2 release was the last one based on 1.0.1 version of the LoRaWAN specification.
The next version 4.4.0 will be based on 1.0.2 version of the LoRaWAN specification.

vikasrohal, 2017-08-02:
Thanks for quick reply, Now working fine.

vikasrohal, 2017-09-11:
Hi, 
My last testing was on Rx window 1. For Rx window 1 this issue closed(working fine), But now facing same issue with Rx window 2.

I am using same code but now Rx window 2  selected from server side(Frequency 8665500000 SF 10 India ISM Band).
I am testing with 10 end node and observer frame counter missed, this is same as issue #287 which has been closed but this time Rx window 2 selected to receive data.

mluis1, 2018-01-16:
Have you been able to solve your issue?
Can we close this issue again?

mluis1, 2018-01-26:
No feedback we close the issue.

closed 2018-01-26 by mluis1