Lora-net/LoRaMac-node #288: EU MAC not respecting duty cycle past the first frame ?

catsup, 2017-08-03:
Hello,

I am experimenting with the latest development branch, to prepare for switching to V4.4 once it is released.

My sample app tries to send an 11 byte payload uplink every 10 seconds at DR0.

I have noticed that for the EU MAC, the duty cycle regulation only seems to be enforced for the first frame - subsequent frames are transmitted without respecting the duty cycle at all, or using random off times.

Is this a known issue ? I'm not including an issue in my port at this stage.

Device side:
```
00:00:00.092193 Entering _tx_task_fn
00:00:00.101501 Entering _send_frame
00:00:00.136779 LoRaMac frame transmission status = 0
00:00:03.812988 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:00:03.826324 unconfirmed, dr 0, txp 1
00:00:03.834136 _mlme_confirm: status 0
00:00:03.841766 Link Check Response, dm20, gw1
00:00:03.850616 _mcps_indication, status 0, McpsInd 0, rxd 0 slot 0 DR0 RSSI -99 SNR 27 port 0 FP 0
00:00:10.099334 Entering _tx_task_fn
00:00:10.108642 Entering _send_frame
00:00:10.131713 LoRaMac frame transmission status = 0
00:00:20.106475 Entering _tx_task_fn
00:00:20.115783 Entering _send_frame
00:00:20.122924 LoRaMac frame transmission status = 1
00:00:30.113616 Entering _tx_task_fn
00:00:30.122924 Entering _send_frame
00:00:30.130065 LoRaMac frame transmission status = 1
00:00:40.120758 Entering _tx_task_fn
00:00:40.130065 Entering _send_frame
00:00:40.137207 LoRaMac frame transmission status = 1
00:00:50.127899 Entering _tx_task_fn
00:00:50.137176 Entering _send_frame
00:00:50.144317 LoRaMac frame transmission status = 1
00:01:00.135040 Entering _tx_task_fn
00:01:00.144378 Entering _send_frame
00:01:00.151519 LoRaMac frame transmission status = 1
00:01:10.142211 Entering _tx_task_fn
00:01:10.151489 Entering _send_frame
00:01:10.158630 LoRaMac frame transmission status = 1
00:01:20.149353 Entering _tx_task_fn
00:01:20.158630 Entering _send_frame
00:01:20.165771 LoRaMac frame transmission status = 1
00:01:30.156494 Entering _tx_task_fn
00:01:30.165832 Entering _send_frame
00:01:30.172943 LoRaMac frame transmission status = 1
00:01:40.163635 Entering _tx_task_fn
00:01:40.172943 Entering _send_frame
00:01:40.180084 LoRaMac frame transmission status = 1
00:01:50.170776 Entering _tx_task_fn
00:01:50.180053 Entering _send_frame
00:01:50.187194 LoRaMac frame transmission status = 1
00:02:00.177917 Entering _tx_task_fn
00:02:00.187225 Entering _send_frame
00:02:00.194366 LoRaMac frame transmission status = 1
00:02:10.185058 Entering _tx_task_fn
00:02:10.194366 Entering _send_frame
00:02:10.201507 LoRaMac frame transmission status = 1
00:02:20.192199 Entering _tx_task_fn
00:02:20.201507 Entering _send_frame
00:02:20.208648 LoRaMac frame transmission status = 1
00:02:30.199310 Entering _tx_task_fn
00:02:30.208648 Entering _send_frame
00:02:30.215698 LCR failed
00:02:30.221008 LoRaMac frame transmission status = 1
00:02:30.521087 Timer func 0xb795 set to 1
00:02:30.527282 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:02:30.540618 unconfirmed, dr 0, txp 1
00:02:30.548431 _mlme_confirm: status 1
00:02:30.556030 Link Check Request FAILED
00:02:30.563903 _mcps_indication, status 10, McpsInd 0, rxd 0 slot 0 DR0 RSSI -97 SNR 29 port 0 FP 0
00:02:40.206451 Entering _tx_task_fn
00:02:40.215759 Entering _send_frame
00:02:40.252166 LoRaMac frame transmission status = 0
00:02:44.419830 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:02:44.433166 unconfirmed, dr 0, txp 1
00:02:44.440979 _mlme_confirm: status 1
00:02:44.448608 Link Check Request FAILED
00:02:44.456481 _mcps_indication, status 10, McpsInd 0, rxd 0 slot 0 DR0 RSSI -97 SNR 34 port 0 FP 0
00:02:50.213592 Entering _tx_task_fn
00:02:50.222869 Entering _send_frame
00:02:50.245819 LoRaMac frame transmission status = 0
00:03:00.220733 Entering _tx_task_fn
00:03:00.230072 Entering _send_frame
00:03:00.237213 LoRaMac frame transmission status = 1
00:03:10.227905 Entering _tx_task_fn
00:03:10.237213 Entering _send_frame
00:03:10.244354 LoRaMac frame transmission status = 1
00:03:18.749023 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:03:18.762359 unconfirmed, dr 0, txp 1
00:03:18.770172 _mlme_confirm: status 3
00:03:18.777801 Link Check Request FAILED
00:03:20.235046 Entering _tx_task_fn
00:03:20.244323 Entering _send_frame
00:03:20.279632 LoRaMac frame transmission status = 0
00:03:24.337799 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:03:24.351135 unconfirmed, dr 0, txp 1
00:03:24.358947 _mlme_confirm: status 4
00:03:24.366577 Link Check Request FAILED
00:03:30.242187 Entering _tx_task_fn
00:03:30.251464 Entering _send_frame
00:03:30.286773 LoRaMac frame transmission status = 0
00:03:33.962921 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:03:33.976257 unconfirmed, dr 0, txp 1
00:03:33.984069 _mlme_confirm: status 1
00:03:33.991699 Link Check Request FAILED
00:03:33.999572 _mcps_indication, status 10, McpsInd 0, rxd 0 slot 0 DR0 RSSI -99 SNR 31 port 0 FP 0
00:03:40.249328 Entering _tx_task_fn
00:03:40.258666 Entering _send_frame
00:03:40.293975 LoRaMac frame transmission status = 0
00:03:44.297851 _mcps_confirm: Status = 0, McpsRequest = 0, DR=0, TXP=1
00:03:44.311187 unconfirmed, dr 0, txp 1
00:03:44.319000 _mlme_confirm: status 1
00:03:44.326599 Link Check Request FAILED
00:03:44.334503 _mcps_indication, status 10, McpsInd 0, rxd 0 slot 0 DR0 RSSI -97 SNR 29 port 0 FP 0
```

Server side:
```
2017-08-03 09:41:50 012ff663 12 868300000     0  data '132f6d3c58782d78583e62'
2017-08-03 09:44:17 012ff663 12 868100000     1  data '132f6d3c58782d78583e62'
2017-08-03 09:44:31 012ff663 12 868500000     2  data '13316d3c58782d78583e62'
2017-08-03 09:45:11 012ff663 12 868100000     4  data '13316d3c58782d78583e62'
2017-08-03 09:45:21 012ff663 12 868300000     5  data '13306d3c58782d78583e62'
2017-08-03 09:45:31 012ff663 12 868300000     6  data '13306d3c58782d78583e62'

```

catsup, 2017-08-03:
... obviously meant to say I'm not EXcluding an issue in my port.

nestorayuso, 2017-08-03:
Maybe your rtc-board.c is not correctly ported.
I had the same issue porting.

catsup, 2017-08-03:
Possibly, yes - thanks for the hint. I'm on an EFM32 processor and don't use the ST-specific RTC code, but I'll double check the timer related code. This same code works fine with the pre-4.0 MAC, but I might have missed something in the microsecond to millisecond unit conversion. Will report back here.

nestorayuso, 2017-08-04:
Me too with EFM32. Check this two repos:
https://github.com/imec-idlab/LoRaMac-node/tree/master for a working  rtc-board.c
https://github.com/DRAMCO/LoRaWAN_EFM32 for Hw acelerated encryption

catsup, 2017-08-04:
Thanks for the pointers - however, I'm only using the new mac/radio and timer code, as my board peripherals (ie, RTC) are used for different purposes than in the Semtech/Stackforce examples.

I've just found the issue though - my rtc-board equivalent was missing one conversion between microseconds / milliseconds.

Closing this issue. Thanks for your help !

closed 2017-08-04 by catsup