Lora-net/LoRaMac-node #290: Just sent first packet of data, Please help !

artuguy, 2017-08-07:
Hi, Everyone:
     I am porting SX1276 and Lorawan stack to stm32l051c8t6, using 433 BAND, after sent a UNCONFIRMED packet, the LoRaMacStatus is always BUSY, and  LoRaMacState always is LORAMAC_TX_RUNNING .
     I found someone have the same  problem, but I can't find the solution, Please help.

THANKS.

djaeckle, 2017-08-08:
Hi artuguy,

how long does the LoRaMAC stay in this state? Do you get a confirmation - if so, please provide the details. Which version are you using?

artuguy, 2017-08-08:
djaeckle 👍 
    Yesterday, I plug the DIO1 to the wrong PIN.  today , I set it to the right position, but reguard  the TxNextPacketTimer , if I set it value <= 3000 , it can enter to IRQ, IF i set it  to TxDutyCycleTime about
4xxx , it can't enter to IRQ.  I can't get a confirmation,  but base station can recieve the data packet , it is right.  I use APB mode, stm32l0538 discovery,  version is 4.3.2. 
     I have no lowpower mode ,so the RTC IRQ is like 

void RTC_IRQHandler( void )
{
    HAL_RTC_AlarmIRQHandler( &RtcHandle );
    HAL_RTC_DeactivateAlarm( &RtcHandle, RTC_ALARM_A );
//    RtcRecoverMcuStatus( );
//    RtcComputeWakeUpTime( );
//    BlockLowPowerDuringTask( false );
    TimerIrqHandler( );
}

artuguy, 2017-08-08:
djaeckle,
      I fallback my code to 2 days ago, and enlarge Led1 timerEvent value from 25 to 2000, I find  after 1st data packet being sent,  a confirmation could be get.

artuguy, 2017-08-08:
djaeckle,
           application sent data duty cycle is 5s, but Is Lorawan MAC dutycycle 2 minite ?  I can send multi data packet , just dutycycle is 2 minite.
           thanks

djaeckle, 2017-08-08:
Hi artuguy,

I think your issue is related to the duty cycle. Please check issue #277.

closed 2018-01-16 by mluis1