Lora-net/LoRaMac-node #293: How to set the wake up time?

suyiii, 2017-08-08:
Hello! I wanna wake up the cpu use the RTC. But I have not found any defination or function to set the wake up time. It must be that I lost something. Could you tell me what I lost? Thank you!

djaeckle, 2017-08-08:
Hi suyiii,

could you please provide a more detailed description? Do you mean the time overhead when your MCU transits from the low power mode into the active mode?

suyiii, 2017-08-08:
No , not that complicate. I didn't describe it correctly.

I wanna set the period from the function HAL_PWR_EnterSTOPMode( PWR_LOWPOWERREGULATOR_ON, PWR_STOPENTRY_WFI ) to the function RtcRecoverMcuStatus() called by RTC not by EXTI. 

I tried to debug the code later. Is it waked up by RTC after any timer is timeout?

djaeckle, 2017-08-08:
Hi suyiii,

thanks for the feedback. Unfortunately, there is no definition for the sleeping period. Currently, the timer implementation is responsible to set the node into sleep mode with `TimerLowPowerHandler`. The RTC will wake up the node in case of an interrupt.

However, the applications in this project are only sample applications. You could develop your own application according to the needs of your custom project, where you may change the low power behaviour to utilize a dedicated sleep period (e.g. between uplink frames).

temporaryaccount, 2017-08-08:
Hello suyiii,

You can also configure and start a Timer object before going to sleep. It will wake up the CPU using an RTC Alarm.

```C
    static TimerEvent_t sleep_timer = { 0 };
    static TimerTime_t sleep_time = 1U * 60U * 1000U; /* 1 hour */

    static void on_sleep_timer(void);

    int main()
    {
        //...
        //...
        
        TimerInit( &sleep_timer, on_sleep_timer, NULL );
        TimerSetValue( &sleep_timer, sleep_time );

        //...
        //...

        case DEVICE_STATE_SLEEP:
        {
            // Wake up through events
            TimerReset( &sleep_timer );
            TimerLowPowerHandler( );
            //...
            //...
        }

        //...
        //...
    }

    static void on_sleep_timer( void )
    {
        return;
    }
```

suyiii, 2017-08-21:
Thank you for your comments! I forgot to close the question.

I have solve the question with your help. I will close the question.

closed 2017-08-21 by suyiii