Lora-net/LoRaMac-node #294: How can i get the distance 1-2Km with SX1276 module

talahata, 2017-08-08:
I use SX1276 module for my node device. My issue is maximum distance that i can get is only about 200m while i sent data with maximum power and SF12. My gateway used SX1301 module.

Can anybody tell me why my distance is so short? And how to reach the distance as the manufacturer put forward?

jamesl-dm, 2017-08-08:
Have you checked your antenna switch and transmit path components?
You should be getting RSSI's of between -60 and -40 dBm with your gateway in the same office as the node. If you're getting closer to -100, your antenna is probably not connected.

jamesl-dm, 2017-08-08:
If you are in a region with lots of channels like AU915 or US915, with ADR turned off so that the gateway doesn't send a channel list back down, your node might also be transmitting on the wrong frequencies. Sometimes a transmission on the wrong frequency will alias into one of the correct frequencies, allowing reception with a greatly attenuated signal strength (~70 dB). So make sure you've set your channel masks if not using ADR.

talahata, 2017-08-08:
Maybe my problem is because of antenna. Currently, i draw ant for my end device as a wire on board, so maybe that is reason for my problem.
Can you gave me some advices to design or use an correct antenna?

jamesl-dm, 2017-08-08:
That's a bit outside my area of expertise. If you google around a bit there are a fair number of application notes with free PCB antenna designs. Inverted F designs are popular at 868 / 915 MHz.
At any rate, even with a horrible antenna, 200 m seems a bit short...

talahata, 2017-08-08:
okie, thanks for your help!

mluis1, 2018-01-16:
No more activity we close the issue.

closed 2018-01-16 by mluis1