Lora-net/LoRaMac-node #296: Any plans to add a flag to prevent delayed transmissions (EU868 duty cycle) ?

catsup, 2017-08-08:
Hi,

Currently, when passing frames to the MAC for transmission on a duty cycled MAC (ie, EU868), the first frame gets sent immediately, a second frame queued during the backoff interval gets delayed and sent later by the MAC, subsequent frames are rejected until the delayed frame has been sent.

For my application, I'd rather not have frames buffered by the MAC as my data gets stale quickly - if a frame cannot be sent right away, I'd prefer an error to be returned immediately so I can discard the frame and send a new one with updated data later on.

Any plans for something like this ?

Thanks,
Marc

nestorayuso, 2017-08-08:
before sending, call to LoRaMacQueryTxPossible()

catsup, 2017-08-08:
That only tells me whether my data will fit in the next uplink frame, not whether I can transmit it right now.
By transmitting, I mean actually sending it on the air - I don't want it to be delayed by the MAC.

Testing whether the a transmission is possible would be an alternative, but that doesn't exist either - see #233

djaeckle, 2017-08-08:
Hi catsup,

thanks for your report. We will discuss your proposition and will come back to you. Please note that it might take some time due to the summer break.

GuntherSchulz, 2017-08-24:
Hi @catsup @djaeckle 

see my comments on #233

amouiche, 2017-12-05:
Hi all,

Could we suggest to add a "TimerTime_t RemainTimeOff" (or something 
like) to the LoRaMacTxInfo_t struct, and have
LoRaMacQueryTxPossible() fill this field.
zero, will mean that the desire packet can be sent straight.
something else tells the minimum time to wait until before next 
immediate Tx.

Regards,
Arnaud

closed 2018-01-26 by mluis1