Lora-net/LoRaMac-node #297: const correctness

tibogens, 2017-08-09:
Hi,

Most functions of the library do not use the `const` qualifier for their arguments, even when it would be desirable. (see [the earch results for 'const' in the src/mac directory](https://github.com/search?utf8=%E2%9C%93&q=const+repo%3ALora-net%2FLoRaMac-node+path%3Asrc%2Fmac&type=Code&ref=advsearch&l=&l=)).

For instance, the `MibRequestConfirm_t*` argument to `LoRaMacMibSetRequestConfirm()` is a good candidate. The argument is never modified by current code (indeed, it's a setter), yet it requires the pointer to be non-const. This forces the user to either use ugly (and/or dangerous) casts, or to have their own functions (e.g. `my_fancy_app_get_nwkskey()` higher-level function) return non-const pointers (quite dangerous in both cases if this data comes straight from the EEPROM).

I've already tried to modify my own copy by adding `const` qualifiers, starting with `LoRaMacMibSetRequestConfirm`. Of course the change creeps down/up in many places of the LoRaMac library, forcing me to add more `const` to many functions and/or structs. This isn't a problem in itself, but I wish to know the following before going on:

- Is there a reason I did not understand that justifies the absence of `const` ? Or has this been simply overlooked?
- If the answer is no, would you be willing to accept a pull request to enforce const correctness? If yes, which branch should I work from?

I can dedicate as much time as needed to implement this, although I do not have the devices to run the application - only my device on which I run the library. 

Cheers,
Thibault

catsup, 2017-08-09:
I second this one. Making constants const also allows for substantial RAM (as well as time) savings.

`LoRaMacInitialization()` is another obvious candidate for const parameters.

mluis1, 2018-01-16:
In general we agree with your statement.

You can always propose your changes against the develop branch. After review we will decide if it is accepted as is or if some changes are still required.

closed 2018-01-16 by mluis1

tibogens, 2018-01-16:
All right - I hope I can find some time to create the PR.