Lora-net/LoRaMac-node #299: OTAA Join Data Rate

RakshitAdmar, 2017-08-10:
I'm using an im880b configured to OTAA mode. The join procedure works well but defaults to data rate SF7. 
The state 'DEVICE_STATE_JOIN` isn't a recurring state if join was unsuccessful. 
Instead the loop goes to `DEVICE_STATE_SLEEP` and doesn't reach `DEVICE_STATE_JOIN` again

Also every time `LoRaMacMlmeRequest` for a join request is executed, the variable `JoinRequestTrials` is reset.

Because of the above, all my join requests always are of datarate SF7.

Do i require a fix or am I doing something wrong?

djaeckle, 2017-08-10:
Hi RakshitAdmar,

thanks for your report. Which version and which region are you using?

RakshitAdmar, 2017-08-10:
I'm on the latest version. Version 4.3.2
I've configured the node to the European 868Mhz band.

djaeckle, 2017-08-11:
Thanks for the input. In OTAA mode it is possible to specify the number of trials (with some limitations with respect to different regions) for the join request.

> The join procedure works well but defaults to data rate SF7.

Could you please specify this statement? Do you means that the device performs join requests always at SF7 or that after a join the node will use always SF7 for the next uplink?

> The state DEVICE_STATE_JOIN isn't a recurring state if join was unsuccessful.
> Instead the loop goes to DEVICE_STATE_SLEEP and doesn't reach DEVICE_STATE_JOIN again

The MAC notifies the application with the MlmeConfirm primitive about the status of the join. In case of an unsucessful join, the MAC should provide a `mlmeConfirm->Status` which is unequal to `LORAMAC_EVENT_INFO_STATUS_OK`. In the sample app, this is also the location where the app sets the device state to `DEVICE_STATE_JOIN` again. 
Regarding to your issue, does the MAC call the MlmeConfirm primitive?

> Also every time LoRaMacMlmeRequest for a join request is executed, the variable JoinRequestTrials is reset.

Yes, the variable will be reset with every 'new' join request.

RakshitAdmar, 2017-08-11:
> The join procedure works well but defaults to data rate SF7.
I mean to say the join request is always at SF7.

Thanks for pointing the recurrence of DEVICE_STATE_JOIN

The mac does call the MlmeConfirm primitive. Is there a specific counter/state you want me to check when it executes this primitive?

My doubt is 
In LoRaMac.c:4334 `LoRaMacParams.ChannelsDatarate = AlternateDatarate( JoinRequestTrials + 1 );`
I've set a breakpoint there and checked that every time this statement is executed, JoinRequestTrials is always 0. Because of the initialization in line 4325.

djaeckle, 2017-08-11:
Thanks for the input. If the join fails, the MlmeConfirm primitive should setup the `DeviceState` to `DEVICE_STATE_JOIN`. In that case the application should restart the join procedure.

If you perform a `LoRaMacMlmeRequest `, the MAC will reset variable `JoinRequestTrials `. You can specify the number of trials for the join request. The MAC will perform the amount of join request you have specified (for EU868, the minimum is 48), before it will provide the MlmeConfirm primitive. During this procedure, the MAC will increase `JoinRequestTrials ` with every join it performs.

Please take a look at the implementation of `AlternateDatarate`. The function specifies, which datarate the MAC will use with respect to the value of `JoinRequestTrials `. So you should see a datarate change if the MAC performs the 48 join trials.

RakshitAdmar, 2017-08-14:
Thank you for pointing me where the `DEVICE_STATE_JOIN` state is reset and more about `JoinRequestTrials`. I was confused between request trials and request procedure. 

I did take a look at `AlternateDatarate` function. It seems that the data rate changes only if the `JoinRequestTrials` is either `{48, 32, 24, 16, 8}`. So amongst the 48 tries, the data rate is changed from SF7 only 5 times. These datarates aren't retried for every join procedure.  
I'm not sure why the implementation is like this, but if it's concerning backoff time, then isn't clogging SF7 for join procedures equally bad ?

djaeckle, 2017-09-06:
Hi RakshitAdmar,

thanks for the feedback. We have received the implementation of the data rate alternation from the technical committee of the LoRaWAN Alliance. It is correct that SF7 will be used multiple times in a row. However, in summary, the implementation should ensure that the 1% duty cycle is respected all the time. The other advantage is that the will always connect at the highest possible data rate.

hasnainvirk, 2017-09-29:
@djaeckle It doesn't make sense. The number 48 doesn't make sense to me. All it needs to do is to pick a join channel and try all the data rates starting from the highest , down to the lowest. Because higher the data rate, lesser is the distance between base station and node. 
At the moment, in European band there are 3 join channels and its mandatory that all the join channels  must be supported by base stations. Now, let's say if at channel 868.30 you sent a Join request with DR_5, next there would be a try on a join channel (randomly selected join channel) with a weirdly calculated DR. I don't know what 48 is. All it needs to do is to lower the data rate with every try until the node finds a base station. But it doesn't work that way in the code.

rraks, 2017-09-29:
Agreed that 1% duty cycle has to be respected. But you could try all the data rates first and then enforce duty cycle limitations. Because in the case of a person performing deployment, ensuring that the device has connected to the network will be a very time consuming task. But i believe we can modify your code to suite our need adhering to the duty cycle limitations, which is what I've done. 
Thanks for your response @djaeckle .

mluis1, 2018-01-16:
We have recently changed the way that the JoinReq are being handled. Please refer to the following commits:
bec887dd2414b091fe62277ed57d0166f77a5055
42cab8c35f0bd1299d8796c2606867030bc226a9
531b4f7318e8d30681573a5198f27e491db086ea
cff4c70a309218075984e1043ef9f2d942dad793
96faff32586a1b8acd663751b3d0fd176ad4f869

Can we close this issue?

hasnainvirk, 2018-01-17:
So are you leaving out the Join request re-transmission algorithm for upper layers to implement ?

mluis1, 2018-01-17:
Yes, it is up to upper layer to retry to send the JoinReq again when it fails.

closed 2018-01-26 by mluis1