Lora-net/LoRaMac-node #300: Sync Node RTC to network server[Question]

RakshitAdmar, 2017-08-11:
I was wondering what the right way to sync the RTC calendar to the network server time is. Is there a best practice you would recommend? I couldn't find documentation about this on the LoRaWAN specs doc for Class C and Class A.

djaeckle, 2017-08-15:
Hi RakshitAdmar,

the LoRaWAN Specification defines with Class B a beacon synchronization. The beacon frame contains a time reference. So with Class B you could synchronize your end device time with this time reference.

RakshitAdmar, 2017-08-15:
My question is, can this be done with Class A and Class C ?
Otherwise, I will have to add the timestamp as payload on the downlink. The downlink is subject to being queued so for class A this method will not work. 

Are you suggesting that I configure my device as a Class B device and then change classes once synchronization is done?

nestorayuso, 2017-08-17:
I think LoRaWAN release 1.1 will have a MAC command DeviceTimeReq for that

GuntherSchulz, 2017-08-21:
Your Gateway / Network provider should have an API giving the exact network time stamp of when a packet was received.
What I do is send an upstream frame containing the time as my end device has it.
The backend then compares that time in the payload to the network timestamp of reception at the gateway.
Now queue a downstream packet containing the SIGNED DIFFERENCE between the two times, and adjust the node's time accordingly.
This way, queing of packets has no bearing on SIGNED TIME DIFFERENCES during previous upstream packet, and time can be adjusted accurately to within the larger resolution of [Network TimeStamp, EndDevice TimeStamp] - 1 second in my case.
Should be more than good enough!

mluis1, 2018-01-16:
Can we close this issue?

GuntherSchulz, 2018-01-17:
I dont mind, but @rraks asked the original question.

mluis1, 2018-01-26:
As said by @nestorayuso the LoRaWAN 1.0.3 and 1.1 have a new MAC command named DeviceTimeReq which can be used for this.

closed 2018-01-26 by mluis1

GuntherSchulz, 2018-11-28:
Great to see my idea formalized :)