Lora-net/LoRaMac-node #301: Class B -  Beacon Acquisition MLME Confirmation callback before Acquisition completes

snelson-senet, 2017-08-15:
Hello, 

I've been playing around with the Class B application and I believe there is currently an issue where the Acquisition MLME confirmation callback runs before the acquisition process has finished and as a result a successful acquisition is not signaled to the application.

The sequence of events I observe are:
1. Successful Beacon Timing Request and Answer
2. MLME_BEACON_ACQUISITION request,  status = OK
3. Application frame transmit
4. At the end of the RX2 window the MlmeConfirm Acquisition callback is triggered with status = 
    RX2_TIMEOUT
5. Application returns to Beacon Timing state on perceived acquistion failure
6 Successful Beacon Timing Request and Answer
7. MLME_BEACON_ACQUISTION requests now fails with status = BUSY
8. Class B algorithm wakes up at the expected beacon time, beacon is received,
    mlme beacon indication callback occurs with status = BEACON_LOCKED, no confirmation callback 
   since  there is no longer an acquistion request on the mlme confirm queue 

At #4, when the confirm callback is triggered, relevant beacon context acquisition state is:
           AcquisitionPending = 0
           AcquisitionTimerSet = 1
           BeaconExpected       = 1
And the acquistion mlme confirm callback is run because  LoRaMacClassBIsAcquisitionPending( ) != true.

From what I can tell, the acquisition pending flag is set only while the radio is in receive mode during the expected beacon receive time.  Any other mlme request before the beacon rx window is opened will result in the acquisition mlme confirmation being prematurely popped off the queue and prevent subsequent acquisitions requests until the acquistion process finishes.

I changed the test to run the acquisition confirm status callback to be when the both acquisition pending and timer flags are not set. With this change I now get the acquistion confirmation
status when the acquistion process completes.

djaeckle, 2017-08-16:
Hi snelson-senet,

thank you very much for your report. I think the issue is that you start a frame transmission during the MAC tries to acquire the beacon with the `MLME_BEACON_ACQUISITION` procedure. What happens, if you wait with the frame uplink until the MAC provides the MlmeConfirm primitive for the acquisition?

However, I think it should be possible to perform class A uplinks during this procedure - even if it results in an acquisition fail due to an uplink/beacon collusion. I will try to reproduce your issue on my side as soon as possible.

snelson-senet, 2017-08-17:
Thanks for the response. The frame transmit request is the root cause of this issue, so pending it until after acquisition completion and status callback removes the failure cause.  

It makes perfect sense preventing frame transmits near the next beacon time, so I would suggest failing a transmit request with a non-ambiguous status indicating beacon acquisition in progress.

djaeckle, 2017-10-09:
Hi snelson-senet,

we have this issue on the roadmap for the class b feature. Please close this thread if it has resolved your issue.

closed 2018-01-26 by mluis1