Lora-net/LoRaMac-node #303: stack implementation LoRa ( stm32L082)

KarimelQ, 2017-08-18:
Hello everyone, 
I'm working on a LoRa project, using stm32l082 & semtech's SX1276, the example program given in I-CUBE-LRWAN 1.1.0 is for STM32L053. Exept this example i have no idea from where i should start, i want to know what are modifications that i should do in order to send uplink data ( from end point to gateway), or an example program to start with.
Thank you

temporaryaccount, 2017-08-23:
> i have no idea from where i should start

Please make sure you have read all the available documentation about [STM32CubeMX](http://www.st.com/en/development-tools/stm32cubemx.html) and [I-CUBE-LRWAN](http://www.st.com/en/embedded-software/i-cube-lrwan.html).

You may get better support for I-CUBE-LRWAN at ST's [community forums](https://community.st.com/community/stm32-community/stm32-forum) than here as the I-CUBE-LRWAN is being developed by them.

KarimelQ, 2017-08-23:
thank you for the advice @temporaryaccount

fahad80, 2017-09-01:
@KarimelQ : I-CUBE-LRWAN 1.1.1 has End_Node example. Start from there.

closed 2018-01-16 by mluis1