Lora-net/LoRaMac-node #304: dutyCycleTimeOff computation on LORAMAC_TX_DELAYED

mbracq, 2017-08-18:
Hi all,
I stress the loramac on last version of the develop branch. I send a lot a frame on an short time (not a real use case).
The issue is  when we do LoRaMacMcpsRequest, sometimes the function ScheduleTx computes a dutyCycleTimeOff different to 0 and delay the frame. It's Ok.
But when TxDelayedTimer expires, sometimes, I found a dutyCycleTimeOff different.

For exemple dutyCycleTimeOff = 9000 ms on first computation and 58000ms on second computation.

I don't think that is a nominal behavior; can you confirm that please?
Maybe we re-add the lastTx a second time on OnTxDelayedTimerEvent. 

Best regards, 
Manuel

djaeckle, 2017-08-24:
Hi mbracq,

thanks for the report. Could you please provide some more information? Which version are you using? Are you using the sample application?

djaeckle, 2017-10-09:
Any news here? Can we close the issue?

mluis1, 2018-01-16:
No activity we close the issue.

closed 2018-01-16 by mluis1