Lora-net/LoRaMac-node #306: Support for FSK/OOK Modem

heroichornet, 2017-08-21:
I want to transmit wireless m-bus. Is this possible with this stack?

djaeckle, 2017-08-21:
Hi heroichornet,

no this is a LoRaWAN stack and not a Wireless M-Bus stack.

heroichornet, 2017-08-21:
Djaeckle, thank you for the quick answer.

Is there a Wireless M-Bus Stack for the SX127x?

nestorayuso, 2017-08-21:
yes, but not opensource.

ping http://www.stackforce.de for a licensing stack or http://www.amihotechnology.com/ for modules with  Wireless M-Bus.

davidrahusen, 2017-08-21:
Yes, of course it's possible. It'll require some effort to get there, but I guess it'll be fairly easy. And if it suits your business case, let's get together to discuss the details! Please drop a mail at info@stackforce.de providing your contact details and I'll get back to you.

closed 2017-08-22 by heroichornet

GitTibbe, 2020-02-17:
I'm trying to send WMBus FSK messages according to the requirements, but nothing is coming through a WMBus router I bought. Except from message formats, what settings are important?