Lora-net/LoRaMac-node #307: node stop after about 4294360 seconds from boot

ASCOEL, 2017-08-22:
Hi All,

I have develop a sensor based on IM880 module and library rel 4.3.2
All the sensors (about 200pcs installed in field) they are stop after about 4294360 seconds.
This value is showed on serial shell available on the sensor, and only the timer is stuck; the shell and other activity not timer-dependent still running fine.
Because the internal timer is express in mS, the value 4294360 is very near to 0xFFFFFFFF mS (4294967296 decimal).
My application reload every 20 seconds the timer for wake-up the processor and perform some activity, at end of activities the processor return in sleep mode awaiting the new timer interrupt within 20 seconds.
After 4294360 seconds the timer is not able to generte a new interrupt and the node remain in sleep mode.
This issue is very strange and is causing of a lot of problems at the customer and network provider (a very important worldwide provider)
Do you have an idea because this happen?
With the library release 3.x.x and the timer expressed in uS this problem was never happened.
thanks and regards
Jerry

temporaryaccount, 2017-08-23:
Duplicate, see #218.

ASCOEL, 2017-08-23:
Ok, I already know the whole problem, but the MASTER development team is not thinking about fixing the problem?
It's a big problem.

mluis1, 2018-01-16:
A fix has been provided. Please refer to issue #218

closed 2018-01-16 by mluis1