Lora-net/LoRaMac-node #309: AS923

coljander, 2017-08-23:
Hi

I'm having trouble getting this working.  I'm trying to connect to a Multitech gateway configured as a network server.  I can see the JoinRequest is reaching the server and that the server seems to be sending back the JoinAccept.

I've checked the code and everything seems to be getting set up correctly regarding the datarate and frequency.  I've checked on a scope (see attachment) and everything seems to be happening at the correct time.  The blue trace shows my 2 rx wondows and the red trace is the pulse coming out of the Multitecth.  So you can see Rx1 and the pulse coincide.  However the RxWindow is closing in the RxTimeout routine.

Any ideas of why this isn't working.  My initial thoughts was that the rxwindow is set up for the incorrect datarate.  Both Rx1 and Rx2 are set up for DR2 but you can see that the Rx windows are open for different times.  However when I check what is getting set they are both getting set with the same parameters.

Any ideas?

Thanks
[image]

finnoliver, 2017-08-23:
Did you set the gateway to Public? http://www.multitech.net/developer/software/lora/conduit-aep-lora-communication/aep-lora-use-third-party-devices/

coljander, 2017-08-23:
Yes.  All other settings are as per the default.

[cid:image001.png@01D31C23.14226D70]

djaeckle, 2017-08-24:
Hi coljander,

I would also recommend to double check the timing. You could also adapt the timing parameters during runtime. The related `MIB `types are `MIB_SYSTEM_MAX_RX_ERROR` and `MIB_MIN_RX_SYMBOLS`. E.g. you could possible increase the `MIB_SYSTEM_MAX_RX_ERROR` for testing.

coljander, 2017-08-24:
Please look at the screenshot I supplied.  You can see quite clearly that the receive windows are open at the same time as the gateway is transmitting, and is open up long enough to receive the expected sync pulse.

I’ve checked the rx configuration and they all seem to be correct.

Also when you compare the 2 receive windows, as they are both waiting for downlinks at the same datarate and for the same amount of sync pulses they should be the same length.  The same values get sent into the stack so why are they different lengths?  I have not changed any of that code.

I’m having no problem with the US, EU and AU code which all uses exactly the same common code, although from memory when I look at the size of the receive windows the first receive window seemed to be longer than I would have expected.

mluis1, 2018-01-16:
No more activity we close the issue.

closed 2018-01-16 by mluis1