Lora-net/LoRaMac-node #310: MAC commands do not get flushed on tx,

catsup, 2017-08-23:
I have an application where I add a link check request to my uplinks, to verify whether the network is reachable. This works fine as long as I am receiving downlink responses, however, in the absence of responses the link check request MAC commands are accumulated in the MAC buffers, resulting in (a) transmission of several link check request commands, and (b) exhaustion of space for user payload, which is particularly annoying in regions where the payload size is limited (ie, US915 and AS923) - these can even get permanently stuck, preventing transmission because the frame exceeds the allowed size - even if no user payload is being sent to try and flush the MAC command buffer.

Is there a way to ensure only one link check mac command is added, or to flush the mac buffers ?

GuntherSchulz, 2017-08-24:
static bool SendFrame( void )
{
    McpsReq_t mcpsReq;
    LoRaMacTxInfo_t txInfo;

    if( LoRaMacQueryTxPossible( AppDataSize, &txInfo ) != LORAMAC_STATUS_OK )
    {
        // Send empty frame in order to flush MAC commands
        mcpsReq.Type = MCPS_UNCONFIRMED;
        mcpsReq.Req.Unconfirmed.fBuffer = NULL;
        mcpsReq.Req.Unconfirmed.fBufferSize = 0;
        mcpsReq.Req.Unconfirmed.Datarate = LORAWAN_DEFAULT_DATARATE;
    }
    else
    {
        if( IsTxConfirmed == false )
        {
            mcpsReq.Type = MCPS_UNCONFIRMED;
            mcpsReq.Req.Unconfirmed.fPort = AppPort;
            mcpsReq.Req.Unconfirmed.fBuffer = AppData;
            mcpsReq.Req.Unconfirmed.fBufferSize = AppDataSize;
            mcpsReq.Req.Unconfirmed.Datarate = LORAWAN_DEFAULT_DATARATE;
        }
        else
        {
            mcpsReq.Type = MCPS_CONFIRMED;
            mcpsReq.Req.Confirmed.fPort = AppPort;
            mcpsReq.Req.Confirmed.fBuffer = AppData;
            mcpsReq.Req.Confirmed.fBufferSize = AppDataSize;
            mcpsReq.Req.Confirmed.NbTrials = 8;
            mcpsReq.Req.Confirmed.Datarate = LORAWAN_DEFAULT_DATARATE;
        }
    }

    if( LoRaMacMcpsRequest( &mcpsReq ) == LORAMAC_STATUS_OK )
    {
        return false;
    }
    return true;
}

catsup, 2017-08-24:
Thanks GuntherSchulz, I get the idea. However sending an empty frame does not appear to flush the mac buffer. (I'm using the devel branch, specifically region AS923 at DR2, in case it matters).

The main point here is about multiple link check requests (or other MAC commands for that matter) being accumulated and wasting valuable payload space.

GuntherSchulz, 2017-08-24:
1. MLME_LINK_CHECKs can only be performed if you have already joined -> can check this with mibReq.Type = MIB_NETWORK_JOINED; LoRaMacMibGetRequestConfirm( &mibReq ); if(mibReq.Param.IsNetworkJoined){ <whatever you were going to do>}

2. Don't stack DUPLICATE MAC commands if possible -> since the only MAC command issuing you have direct control over is MLME_LINK_CHECK, don't issue a new one unless MlmeConfirm reports MLME_LINK_CHECK completion (failure/pass not significant here, only COMPLETION)

3. Try not to queue MLME_LINK_CHECK while an active tx cycle is busy - see 4. below

4. Before calling LoRaMacMlmeRequest(MLME_LINK_CHECK ), note that LoRaMacMlmeRequest only checks if LoRaMacState has LORAMAC_TX_RUNNING, and fails to check if LORAMAC_TX_DELAYED is also set. To work around this, call LoRaMacMcpsRequest with mcpsRequest->Type == MCPS_MULTICAST (unhandled switch case in that function, so no other side effects will occur). Return of LORAMAC_STATUS_BUSY will indicate LoRaMacState is (LORAMAC_TX_RUNNING or LORAMAC_TX_DELAYED), while the only other possible outcome of LORAMAC_EVENT_INFO_STATUS_ERROR will now actually indicate LoRaMacState is NOT (LORAMAC_TX_RUNNING or LORAMAC_TX_DELAYED).

@mluis1 @djaeckle Please confirm this workaround, and perhaps a patch?
@catsup Please let me know if this works for you

GuntherSchulz, 2017-08-24:
p.s. in the fail branches of MlmeConfirm(MLME_LINK_CHECK) and McpsIndication(), i suggest adding
            MibRequestConfirm_t mibReq;
            mibReq.Type = MIB_NETWORK_JOINED;
            mibReq.Param.IsNetworkJoined = false;
            LoRaMacMibSetRequestConfirm( &mibReq );
to set mac state to UNJOINED, then start a re-join process

GuntherSchulz, 2017-08-24:
@catsup 
If you are doing link checks on every packet simply to get delivery confirmation, rather consider sending a confirmed frame (much more lightweight), and only do a link check if confirmation never comes.

It is less time on air for all hardware components in the system (and thus also less power - especially important in battery-operated systems), and especially on a public network, you're essentially robbing other users of valuable, finite available airtime per band/spreading factor.

The only stat you get from link-check that you would not get from a confirmed comm. is the number of gateways that actually heard your request. While useful in an installation process (to find better physical locations etc,), it is largely irrelevant for normal comms for an unmoving device.

Besides, once you have a gateway count, I gaurantee you your service provider isn't going to be putting up more towers every time you send a packet! :)

djaeckle, 2017-08-24:
Hi GuntherSchulz,

> Before calling LoRaMacMlmeRequest(MLME_LINK_CHECK ), note that LoRaMacMlmeRequest only checks if LoRaMacState has LORAMAC_TX_RUNNING, and fails to check if LORAMAC_TX_DELAYED is also set. To work [...]

Thanks for the report. We will double check it.

catsup, 2017-08-28:
Hello,

thanks for all the suggestions. However, this is really about link check requests and the fact that the MAC adds duplicate MAC requests and does not flush them on tx.

I am aware that a link check request uses airtime and device resources, but that's not my concern in this particular use case.

Join status neither - these requests are only sent when joined (and shall not be used as a factor to consider disconnecting).

This is not about delivery confirmation either - I want to know the gateway link margin, on a moving device. Confirmed frames are not an alternative, as (1) I don't care whether a specific frame actually makes it, and don't want the retries associated with it, (2) I want the gateway link margin to fine tune my tx strategy for the next uplink, and (3) I have seen the MAC get stuck with confirmed frames in the past, including once on the devel branch (haven't been able to reproduce/investigate this so far).

LORAMAC_TX_DELAYED is something I'd like to see removed as well (or at least made optional - see #296). In any case, the application layer should not have to have to know/care about LoRaMAC implementation details or internal states, which aren't available anyway outside of the LoRaMac.c source module anyway.

Regards,

djaeckle, 2017-10-09:
Hi everybody,

sorry to the late response. This issue is on the roadmap for the next version:

https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap

closed 2018-01-16 by mluis1