Lora-net/LoRaMac-node #312: Antenna gain clobbered during joins

jamesl-dm, 2017-08-25:
ResetMacParameters() is called during joins, and it resets the antenna gain to the default antenna gain. There is no way of setting the default antenna gain (short of changing the regional #defines). So there is currently no way to transmit a join request at full power if your board has an inefficient antenna, and join requests will be transmitted at additional power if your board uses a directional antenna.

Suggested fix: set the default antenna gain as well in the mib set request.

djaeckle, 2017-09-06:
Hi jamesl-dm,

thanks for the report! We will check it.

djaeckle, 2017-10-09:
Hi jamesl-dm,

this issue is on the roadmap for the next version: https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap

closed 2018-02-05 by mluis1