Lora-net/LoRaMac-node #314: Incorrect maximum payload size in AS923 MAC causes downlinks to be discarded

catsup, 2017-08-28:
Hello,

I am using the AS923 MAC with its default settings, join type ABP. The server uses DR2 in RX window 2 for downlinks.

According to the LoRaWAN Regional Parameter for AS923 (v1.0-20161012, tables 44/45), the downlink MAC payload size is 19 bytes for a DownlinkDwellTime of 1, which I suppose is the MAC default setting. However, in OnRadioRxDone() the MAC (devel branch) checks against a maximum size of 11 bytes, and rejects the downlinks from the server which contain more than 11 bytes of MAC data as shown in the log below:                        
```
00:35:28.527313 OnRxWindow2TimerEvent                                                                                  
00:35:28.926605 OnRadioRxDone                                                                                          
00:35:28.927551 MAC RX: Frame too large, (size 31 - overhead 13) = 18 which is > 11                                    
00:35:28.940002 OnMacStateCheckTimerEvent, LoRaMacState 00000041 flags 13                                              
00:35:28.950103 _mcps_confirm: Status = 0, McpsRequest = 0, DR=2, TXP=0                                                
00:35:28.959899 MCPS_UNCONFIRMED, dr 2, txp 0                                                                          
00:35:28.966278 _mcps_indication, status 1, McpsInd 0 dlc 0, rxd 0 slot 1 DR2 RSSI -45 SNR 33 port 0 FP 0

```
Is my documentation out of date, or is that an issue in the MAC ? (If the latter, this could also explain issue #309)

Thanks,

nestorayuso, 2017-08-29:
an updated regional parametr here
https://www.thethingsnetwork.org/forum/uploads/default/original/2X/d/dca7977549092f2c594a601caa739e1bacbeea59.pdf

catsup, 2017-08-29:
Thanks @nestorayuso - got the updated version.

The tables are the same though, and confirm that the max frame length should be 19 rather than 11 for DR2. In fact, it looks like the limits in the AS923 region header file have been set with the values from the US915 region.

```
/*!
 * Maximum payload with respect to the datarate index. Can operate with and without repeater.
 * The table proides repeater support. The table is only valid for uplinks.
 */
static const uint8_t MaxPayloadOfDatarateDwell1UpAS923[] = { 0, 0, 11, 53, 125, 242, 242, 242 };

/*!
 * Maximum payload with respect to the datarate index. Can operate with and without repeater.
 * The table proides repeater support. The table is only valid for downlinks.
 */
static const uint8_t MaxPayloadOfDatarateDwell1DownAS923[] = { 0, 0, 11, 53, 126, 242, 242, 242 };

```
In addition, the MaxPayloadOfDatarateDwell1UpAS923 table is being checked for downlinks, I suppose it should be MaxPayloadOfDatarateDwell1DownAS923.

Regards,

catsup, 2017-08-29:
I confirm that the AS923 DR2 downlinks (and other datarates too) work fine with the proper max frame length values as in the specs and the changes from @nestorayuso .

Regards,

calebbramwell, 2017-09-06:
We have also come across this problem, however I don't believe this is an issue with the LoRaMAC-node implementation.
Note for AS923 the max payload size listed in the spec is the M payload size not the N payload size.

**AS923 max payload size from sec. 2.7.6 of the 1.0.2 Regional Parameters spec:** 
[image]

**from sec. 4.3.2 of the 1.0.2 LoRaWAN spec:**
[image]

So the max value of N will always be at least 8 bytes less than M. With the presence of the FOpt control field the value of N will be even lower depending on the size of the FOpt field.
The fact that you are receiving 18bytes in the FRMPayload with DownlinkDwellTime=1 on DR2 would be due to an incorrect implementation of the network server.

catsup, 2017-09-06:
Well spotted, thanks @calebbramwell ! I had completely missed that paragraph.

Makes you wonder how a 12-byte join accept can be transmitted while remaining compliant...

Closing this issue.
Thanks

closed 2017-09-06 by catsup