Lora-net/LoRaMac-node #317: battery Level meaning

FoujanetG, 2017-09-01:
Hi all,

What is the meaning of "baterry level"?
Is it the voltage level of the device or is it the energy remaining in the battery?

If it's the voltage level, the measure can be done during a high load ( when there is a LoRa TX for example) or during a low load : the battery level can be different.

the specification 1.0 says: 

> The battery level, 1 being at minimum and
254 being at maximum

In the case of voltage level, the minimum is it 0V or it's the lowest voltage level of battery where the product is able to normally working?

thanks

bhtalexis, 2017-12-06:
Hi FoujanetG,

The battery level is an indication just like on our smartphone. So you have to use the rule of three with 1 corresponding to 0% and 254 to 100%, and effectively the value is for a device able to work.
You could see this value in percent on your LoRa Network Server's hmi.

Regards

mluis1, 2018-01-16:
No more activity we close the issue.

closed 2018-01-16 by mluis1