Lora-net/LoRaMac-node #320: rtc-board.c :: RtcConvertCalendarTickToTimerTime(NULL) goes backwards

ChrisEAlfred, 2017-09-07:
We are using the L07x.

I have observed cases where time goes randomly backwards by a second within a few minutes.

This was first observed by setting a timer to go off every 30s in main.cpp and getting the time using TimerGetCurrentTime().

I added code inside RtcConvertCalendarTickToTimerTime(NULL) to capture the previous result and compare it to the new result, and verified that the time did go backwards.

As this function is also ultimately used by the MAC layer, it concerns me that this may cause false timeouts.

GregCris, 2017-12-21:
This is not something we have observed.
I am sending you here an example of basic unit test which can be used to test the process of the timer implementation. (please change the .log in .c)
[main.log](https://github.com/Lora-net/LoRaMac-node/files/1579427/main.log)

Using the Tx-Cw project , you should be able to probe the timer completely using tests such as:

    while( i < 0xFFFF )
    {
        TimerSetValue( &Led1Timer, i );
        val1 = TimerGetCurrentTime( );
        TimerStart( &Led1Timer );
        
        while( Led1TimerEvent == false );

        Led1TimerEvent = false;

        val2 = TimerGetValue( );
        val3 = TimerGetCurrentTime( );
        val4 = TimerGetElapsedTime( val1 );

        if( val4 > ( i + 1 ) )
        {
            printf(" val4 %u  > i %u  ( val1 %u )\n\r", val4, i, val1 );
        }

        if( ( val2 != val4 ) && ( val2 > ( val4 + 1 ) ) )
        {
            printf("val2 %d, val4 %d \n\r", val2, val4 );
        }

        i++;
    }

mluis1, 2018-01-16:
Can we close this issue?

closed 2018-01-26 by mluis1

ChrisEAlfred, 2018-06-13:
I don't think you ran the test for long enough - you need to catch it at the right time when the milliseconds have rolled, but the seconds have not.

See https://github.com/Lora-net/LoRaMac-node/issues/480
It had to run for about 4 minutes.