Lora-net/LoRaMac-node #328: Decouple MAC layer from microcontroller stuff.

fifteenhex, 2017-09-24:
More of a question than an issue; I've got the MAC part to compile on linux. I'm going to write a bit of wrapper code so that it can talk to an sx1276 via spidev or the sx1276 kernel driver I'm tinkering on.

I got to wondering why is the MAC part of this so tightly integrated with all of the stuff to support different boards etc? I would have used LMIC which is structured a bit more traditionally but it's apparently dead upstream and doesn't support the region I need.

It would be nice if the MAC was more split out. It would be more portable. It could be wired up to unit testing code to feed it a bunch of test cases to make sure it operates correctly etc.

djaeckle, 2017-10-20:
Hi fifteenhex,

thanks for your report. We are aware of this issue and consider this as an important task. Our motivation is to remove dependencies in the future step by step.

mluis1, 2018-01-26:
We have added this task to our internal task tracking system.
For the moment we close this issue.

closed 2018-01-26 by mluis1