Lora-net/LoRaMac-node #329: Downlink Issue

rohitkumar5293, 2017-09-25:
Hi, 

I am newbie to this technology. I have ported this complete LoRaWAN stack for my MCU using SX1276 transceiver via SPI. I am able to do uplink transmission properly but unable to do schedule  downlink. Hence my MCU doesn't contain RTC so i opt out RTC part. Please suggest how to do schedule downlink.

djaeckle, 2017-10-09:
Hi rohitkumar5293,

you need an accurate timer to schedule the downlink windows. If you say you removed the RTC implementation you need another timer to do the job.

rohitkumar5293, 2017-10-10:
Hi Djaeckle,

As you quote above that RTC is necessary for schedule downlink, now i am implementing RTC complete. Now 1 more problem which i am facing is DIO0 pin. DIO0 pin is responsible for making interrupt between TXDONE and RXDONE. But in my case i am implementing DIO0 pin to make interrupt but i am unable to get downlink.

eric780711, 2018-01-03:
Hi rohitkumar5293
  I have the same problem, do you have a solution?

mluis1, 2018-01-26:
Have you found a solution for this issue?

Most certainly the issue is coming from the porting process.

Can we close the issue?

closed 2018-02-05 by mluis1