Lora-net/LoRaMac-node #330: Using the HSI or MSI clock

temporaryaccount, 2017-09-25:
Some LoRa node devices should be really energy-efficient so using one of the HSI or MSI clocks as the main clock shouldn't be something not ordinary.

However this code below suggests that the device can be running on HSI or MSI clock in the RTC interrupt handler only after waking up from a sleep mode. So it was forcing my MSI clocked device to run the MCU initialization code (and reinitialize the SX127x) after every RTC alarm (i.e. timer event):

```C
void RtcRecoverMcuStatus( void )
{
    //...
    // ...

    // check the clk source and set to full speed if we are coming from sleep mode
    if( ( __HAL_RCC_GET_SYSCLK_SOURCE( ) == RCC_SYSCLKSOURCE_STATUS_HSI ) ||
        ( __HAL_RCC_GET_SYSCLK_SOURCE( ) == RCC_SYSCLKSOURCE_STATUS_MSI ) )
    {
        BoardInitMcu( );
    }
}
```

mluis1, 2018-01-26:
As there is no activity, can we close this issue?

temporaryaccount, 2018-01-30:
If you don't think there is a bug here then sure.

closed 2018-02-05 by mluis1