Lora-net/LoRaMac-node #334: Start transmitting only if User Button is pressed

griphi, 2017-09-29:
Hi, 

currently, I am using STM32 Nucleo-64 board with Semtech SX1272 and the Multitech Conduit Gateway. 
I have a problem with my code. If I push the button, the STM32 Board is starting the transmission. But only once. If I try to push the button after 10 sec, the STM32 Board is transmitting nothing to the Gateway.

here my code:

`    BSP_PB_Init(BUTTON_KEY, BUTTON_MODE_GPIO);

    while(BSP_PB_GetState(BUTTON_KEY) != GPIO_PIN_RESET)
    {
      HAL_GPIO_TogglePin(LED2_GPIO_PORT, LED2_PIN);
      /* Insert delay 100 ms */
      HAL_Delay(100);
    }
    
    while(BSP_PB_GetState(BUTTON_KEY) != GPIO_PIN_SET)
    {
      BSP_LED_Off(LED2);
      lora_fsm( );
      DISABLE_IRQ( );
//      /* if an interrupt has occurred after DISABLE_IRQ, it is kept pending 
//      * and cortex will not enter low power anyway  */
      if ( lora_getDeviceState( ) == DEVICE_STATE_SLEEP )
      {
#ifndef LOW_POWER_DISABLE
      LowPower_Handler( );
#endif
      }
      ENABLE_IRQ();
    }`

Thanks

fahad80, 2017-10-10:
How does your device wake up from sleep?

mluis1, 2018-01-16:
No activity we close the issue.

closed 2018-01-16 by mluis1