Lora-net/LoRaMac-node #335: [Question] Why CAD is not being used for carrier sensing/LBT

hasnainvirk, 2017-09-29:
Radio drivers provide channel activity detection mechanism by doing a CAD. Why this feature is not being used for LBT ? Why an attempt is being made to use IsChannelFree() to determine if RSSI is above a certail level ? Couldn't this be achieved by CAD ?

mluis1, 2018-01-16:
The CAD can't be used because it only detects LoRa modulation preambles (it is not able to detect LoRa modulated data). The CAD feature is most useful on point to point communications where the node transmitting sends very long preambles and the receiver side wakes up time to time.

This is the reason why only the RSSI is used for the LBT.

hasnainvirk, 2018-01-17:
Understood. Thanks.

gnob417, 2018-01-28:
@mluis1 
However, it seems StartCad function is not even implemented in sx1272/6 driver source.
why is that? maybe there is a way to improve LBT mechanism using CAD.

mluis1, 2018-02-05:
@gnob417 The drivers do contain a function named `SX1272StartCad` or `SX1276StartCad`.

As previously said the LoRa CAD is of no use for a LoRaWAN end-device implementation.

closed 2018-02-05 by mluis1