Lora-net/LoRaMac-node #338: EU868_BANDx definitions broken

svenKautlenbach, 2017-10-09:
Hi!

I get compilation error in **RegionEU868.c** file where **Bands** array is defined. The **struct Band_t** has 5 items while the macro only defines 4. Based on the comments in code it seems like **LastJoinTxDoneTime** member is missing.

`missing initializer for field 'TimeOff' of 'Band_t {aka struct sBand}' [-Werror=missing-field-initializers]`

[image]

In order to make it compile I added missing initializer values.
[image]

I am using GCC version from "GNU ARM Embedded Toolchain 6-2017-q2-update"  with -Wall and -Werror. This means also I am not utilizing the default project and build setup that is provided in the repo.

nestorayuso, 2017-10-09:
I updated the pull request https://github.com/Lora-net/LoRaMac-node/pull/302 with this issue in all regions

mluis1, 2017-10-10:
@nestorayuso Pull-request has been merged to develop branch.
We close the issue.

closed 2017-10-10 by mluis1