Lora-net/LoRaMac-node #343: Tx channel config vs base station

Regimbal, 2017-10-18:
Hello everyone,
I observed a weird behavior accidentally.

My device is programmed with latest master branch release (v4.4.0), on EU868 band.
Additional channels are untouched on the main.c file (867.x range).

While doing some tests with my multitech Conduit which was configured with default settings, I observed fix things:
- Packets are received in some 869.x channels
- Some packets are missing

Looking into the lora server of the Conduit, I've seen it was configured with 869.x range additional channels instead of 867.x to match my device programming (my mistake).
Once I set up the Conduit with the proper additional channels, every packets are successfully received, on the correct channels (mandatory ones + 867.x additional ones)

What I don't get is why the device was sending packets in the 869.x range while the gateway was set up with 869.x additional ones... My gateway shall be able to receive in this range (+ mandatory channels), but my device shall not send on these additional ones, shall it?

nestorayuso, 2017-10-18:
When your device joins the network with OTAA, the server sends some MAC commands to the device to enable new channels acording to the ones configured in server and gateway.

It is better to disable aditional channel LC4, LC5... in the device with:
`#define USE_SEMTECH_DEFAULT_CHANNEL_LINEUP          0`
and let the server configure the device channels during OTAA join.

Regimbal, 2017-10-18:
Yes indeed I am using OTAA and was expecting such an explanation.
I disabled addition channels as you recommended, set the Conduit as default (i.e. add channel 869.5MHz to handle 869.1 .3 .5 .7 .9 in addition to default ones) and my first results are:

- some Packets are missed (sequence number missed)
- packets are sent to every theoretical channels except 869.3MHz

By the way, I looked at my former logs and the observation is the same (what I mis configured the Conduit vs device). Packets were received from every channel in 868+869 range except on 869.3...

There must be some relation between the not received packets and the channel I have never received any packed from!

nestorayuso, 2017-10-18:
check that your gateway and your network server use the same channel plan (i think you are using the lora server inside the gateway)

Regimbal, 2017-10-18:
Yes, at this stage I use the server embedded within the Conduit, which is set to EU868 with additional channel 869.5 (which is the center channel for 869.1 .3 .5 .7 .9).
I can see on the multitech forum there are questions around the 869.3MHz use (regulations).
That is probably a reason. I don't know how I can look at the precise channels the Conduit can listen on. Will set it to 867.5MHz instead...

reissjason, 2017-10-18:
There is an updated network server available with channels in the upper range changed to 868.8, 869.0, 869.525, 869.8
http://www.multitech.net/developer/downloads/