Lora-net/LoRaMac-node #344: Question: Why the 1ms delay in SX1272Send in sx1272.c?

travisgriggs, 2017-10-19:
Topic mostly asks it. There are a number of (arbitrary?) 1ms delays in the sx1272.c. I have looked and looked at the data sheets, but I do not see why these delays need to be coded in there when transitioning between standby and transmit modes. Is there something in the data sheets I've missed? Is this just an undocumented "feature" of the hardware?

My reason for caring, is that I would like to get rid of it in our own code, if possible. We're doing some comms at SF7 and dealing with congestion issues, so I'd like to use CCA detection to implement "listen before talk". That 1ms delay is a window of opportunity for a valid CCA to become invalid before transmit, so anything I can do to tighten up the transaction between CCA and TX is a win.

Bonus Question: Also, why the delay in SX1272IsChannelFree. Same issue as above.

mluis1, 2018-01-16:
The 1 ms delay is there to let time for the radio to wake up from sleep.
The datasheet describes those timings on chapter "4.2.6. Startup Times"

Please note that we have modified the SX1272IsChannelFree function in order to just wakeup from sleep and then read as fast as possible RSSI values for the given period.

closed 2018-01-16 by mluis1

travisgriggs, 2018-01-18:
Thanks @mluis1 !