Lora-net/LoRaMac-node #345: Class B initial beacon acquisition never ends in OTAA mode

tortizactility, 2017-10-19:
On Release v4.4.0 in branch feature/class_b, I found that the Beacon Acquisition step never stops after a Join Accept (tested on AS923 or US915). Only few uplinks are sent before entering into Beacon Acquisition mode.

In this situation if the device is started in a location without any beacon reception, the code stuck it into a beacon acquisition mode. It will be better to move similar to BEACON_MISSED if there is no beacon and allow uplink/downlink in Class A.  Then, the code can retry the Beacon Acquisition later on if beacon is now present.

djaeckle, 2017-10-20:
Hi tortizactility,

thanks for your report. We are of this issue and we will provide an update of the class b implementation as soon as possible.

tortizactility, 2017-10-20:
Hi Djaeckle,

During my test, the issue is present even if the Join Request was replied after one or multiple attempts.
Anyway I will recheck that and see if the issue is related to a join request retry.

Thanks a lot for your quick feedback

tortizactility, 2017-10-20:
It can be problematic if the beacon channel is customized via the BeaconFreqReq.
The device will stay in beacon acquisition mode based on the LW default channel.
I think after a timer, the device should revert in Class A and send uplink and retry the Beacon Acquisition later on.

The Beacon Acquisition function, is only waiting for beacon and the MAC command received in response of the uplink are not analyzed.
I fixed it temporary by commenting line 797 in LoRaMacClassB.c
`       // Return always true, when we expect a beacon.
        beaconReceived = true;`

tortizactility, 2017-10-20:
Anyway did you know why the usage of BeaconTimingReq/Ans was removed ?
These MAC commands was very useful to know exactly when the beacon is coming during the device startup. (see commit : https://github.com/Lora-net/LoRaMac-node/commit/10f95c0694ee8f66bd0e9cba81bcec14ecf388e3)

mluis1, 2017-10-20:
They have been removed because the LoRaWAN 1.1 specification deprecated them.
The BeaconTimingReq/Ans can be replaced by the new ClassA MAC command DeviceTimeReq/Ans.

tortizactility, 2017-10-23:
Hi mluis1,

I know that BeaconTiminReq/Ans is deprecated (not removed) in LW 1.1 but the current code in develop branch is not LW 1.1 compliant at all so I don't really understand why this MAC commands was removed.

mluis1, 2018-02-05:
We have reverted back the BeaconTimingReq/Ans code.
We are currently implementing the DeviceTimeReq/Ans MAC command.

closed 2018-02-05 by mluis1