Lora-net/LoRaMac-node #350: Unreliable transmissions

simonnilsson, 2017-10-25:
Hi, I have ported this library to the STM32L052 mcu with SX1276 (http://www.risinghf.com/product/rhf76-052) using the MoteII as base. I have got it almost working but I'm experiencing some problems. 

When testing with an uplink-interval of 10 seconds (Dutycycle check disabled) it occasionally fails to transmit a series of packets  (the time until next successful send also varies). What happens when the transmit fails is it ends up at LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT in OnMacStateCheckTimerEvent. This happens in both ABP and OTAA. Also tested with 30 second uplink-interval.

I know this isn't much to go on but I was hoping someone has an idea what the cause of this inconsistency might be. I suspect it might be a problem with the clock configuration but I have tried different configurations without result.

nestorayuso, 2017-10-25:
check IRQ configuration on DIOx pins

simonnilsson, 2017-10-25:
Hi nestorayuso, thanks for your response.
The code used to initialize the DIOx pins is the following:
```c
{
    GpioSetInterrupt( &SX1276.DIO0, IRQ_RISING_EDGE, IRQ_HIGH_PRIORITY, irqHandlers[0] );
    GpioSetInterrupt( &SX1276.DIO1, IRQ_RISING_EDGE, IRQ_HIGH_PRIORITY, irqHandlers[1] );
    GpioSetInterrupt( &SX1276.DIO2, IRQ_RISING_EDGE, IRQ_HIGH_PRIORITY, irqHandlers[2] );
    GpioSetInterrupt( &SX1276.DIO3, IRQ_RISING_EDGE, IRQ_HIGH_PRIORITY, irqHandlers[3] );
}
```

jamesl-dm, 2017-10-26:
I vaguely recall ending up in LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT when a bug in my timer implementation caused some of the timer callbacks to go off out of order, and at the wrong time.

simonnilsson, 2017-10-27:
By updating rtc-board.c/.h and timer.c/.h using the code in #243 seems to have solved the problems. Thanks.

closed 2017-10-27 by simonnilsson