Lora-net/LoRaMac-node #351: RSSI Calculation

skpeddy, 2017-10-25:
From datasheet sx1276_77_78_79.pdf, page 87 says:

> The same formula can be re-used to evaluate the signal strength of the received packet:
>  Packet Strength (dBm) = -157 + Rssi, (when using the High Frequency (HF) port)
>  or
>  Packet Strength (dBm) = -164 + Rssi, (when using the Low Frequency (LF) port)

In sx1276.c driver code, the calculation goes as:
```
if (SX1276.Settings.Channel > RF_MID_BAND_THRESH)
{
	SX1276.Settings.LoRaPacketHandler.RssiValue = RSSI_OFFSET_HF
						+ rssi + (rssi >> 4);
}
else
{
	SX1276.Settings.LoRaPacketHandler.RssiValue = RSSI_OFFSET_LF
					+ rssi + (rssi >> 4);
}
```

Why is this additional offset (**rssi >> 4**) which is not mentioned in datasheet? 
Thanks in advance.

skpeddy, 2017-10-25:
Seems I am be able to relate RSSI calculation in source code to following equation from data sheet of sx1276.
> RSSI = -157+16/15 * PacketRssi (or RSSI = -164+16/15 * PacketRssi)   

16/15 * PacketRssi approximates to 17/16 *rssi in source code.

Sorry for asking the question. If this understanding is not correct, please correct.

mluis1, 2018-01-16:
Can we close this issue?

closed 2018-02-02 by mluis1