Lora-net/LoRaMac-node #352: Some Question about dutyCycleTimeOff in LoRaMac.c

suyiii, 2017-10-31:
Hello!

I want to control the dutyCycleTimeOff to send the join request frames because the time interval doesn't  meet our requirements.

I use V4.3.2.

I print out the value of TxTimeOnAir and dutyCycleTimeOff and I found that the two variable quantity would be influenced by datarate. DutyCycleTimeOff will be set to 141275 when the datarate is SF12. It means that the program needs to send the next join request frame after more than 2 minutes. 

DutyCycleTimeOff is calculated by complex algorithms, and it seems that I can't set the dutyCycleTimeOff in the function ScheduleTx directly. What can I do to handle the problem?

Thank you for your help!

zhang816kui, 2017-11-09:
you can shut down the compute of  dutyCycle when the system is joinning by modify the CalculateBackOff function,we did that.

heroichornet, 2017-11-17:
@zhang816kui  how did you modify it? With dummy minimum values?

suyiii, 2017-11-23:
@heroichornet, I followed @zhang816kui 's suggestions. I modifed JoinDutyCycle function to return a fixed value. If using BACKOFF_DC_1_HOUR, the interval will become 15 seconds.

Thank you for your help!

closed 2017-11-23 by suyiii