Lora-net/LoRaMac-node #353: Incorrect return code from RegionVerify(PHY_NB_JOIN_TRIALS) -> 48 OTAA join requests.

catsup, 2017-10-31:
Hello,

I am seeing an unexpected number of retries when attempting an OTAA join (Release 4.4, EU868). I am requesting a single join trial, like so:
>	MlmeReq_t mlmeReq;
>
>	mlmeReq.Type = MLME_JOIN;
>
>	mlmeReq.Req.Join.DevEui = lora_settings->deveui;
>	mlmeReq.Req.Join.AppEui = lora_settings->appeui;
>	mlmeReq.Req.Join.AppKey = lora_settings->appkey;
>	**mlmeReq.Req.Join.NbTrials = 1;**
>	rc = LoRaMacMlmeRequest(&mlmeReq);

However, the MAC sends 48 consecutive Join Requests - without regards for EU868 duty cycle limitations it seems.

This appears to be due to an incorrect test in the region code. *LoRaMacMlmeRequest()* checks that the number of trials is valid by calling *RegionVerify()*, and provides a default value of *48* if not.
 
>            // Verify the parameter NbTrials for the join procedure
>            verify.NbJoinTrials = mlmeRequest->Req.Join.NbTrials;
>
>            if( RegionVerify( LoRaMacRegion, &verify, PHY_NB_JOIN_TRIALS ) == false )
>            {
>                // Value not supported, get default
>                getPhy.Attribute = PHY_DEF_NB_JOIN_TRIALS;
>                phyParam = RegionGetPhyParam( LoRaMacRegion, &getPhy );
>                mlmeRequest->Req.Join.NbTrials = ( uint8_t ) phyParam.Value;
>            }

*RegionVerify()* dispatches to region specific verification code, in my case *RegionEU868Verify()*, which compares the passed value against some literal value, and is expected to return *true* if the parameter is valid.
>        case PHY_NB_JOIN_TRIALS:
>        {
>            if( verify->NbJoinTrials < 48 )
>            {
>                return false;
>            }
>            break;
>        }

Unless I'm mistaken, this is wrong. The same problem exists in several other regional implementations.

Bonus question - where does the value *48* (or *2* for some regions) come from ? I can't seem to find this value in the regional specs.

Thanks,
Marc

heroichornet, 2017-11-14:
I've also noticed this, and posted a duplicate issue. I will close mine. I believe this needs to be addressed. I want to know the answer to the bonus question as well :+1:

closed 2018-01-16 by mluis1

catsup, 2018-01-16:
Thanks for the update.

So if I get this right, it is now up to the application to handle the retries, as well as taking care of any region specific duty cycle limitations and datarate selections for subsequent join attempts ? If so, that seems like a step back to me, as the region-specific stuff is (imho) much better handled by the MAC.

mluis1, 2018-01-16:
The duty-cycle is still managed by the MAC layer and the regions sub layer. 

What happens now is that when one tries to send a JoinReq and the MAC can't send it due to the duty-cycle, the LoRaMacMlemRequest function will return LORAMAC_STATUS_DUTYCYCLE_RESTRICTED.
When this happens it is then up to the application to call again the LoRaMacMlemRequest up until the LORAMAC_STATUS_OK is returned.

Please take a look at the updated main.c example files.

catsup, 2018-01-16:
Thanks mluis. I wasn't sure what was happening to the duty cycle with your changes.

Our approach is quite different to the github examples, but I get the idea.

catsup, 2018-01-17:
I just took a look at the code (devel branch), and am somewhat confused about what is being done.

The LoRaMacMlmeRequest() does not return LORAMAC_STATUS_DUTYCYCLE_RESTRICTED, but LORAMAC_STATUS_SUCCESS, the substitution being done in ScheduleTx():

``
    case LORAMAC_STATUS_DUTYCYCLE_RESTRICTED:

        if( dutyCycleTimeOff != 0 )
        {
            // Send later - prepare timer
            LoRaMacState |= LORAMAC_TX_DELAYED;
            TimerSetValue( &TxDelayedTimer, dutyCycleTimeOff );
            TimerStart( &TxDelayedTimer );
        }
        return LORAMAC_STATUS_OK;
``

I thought the onus was now on the application to do the retries. Shouldn't the above case be changed to return the matching status code, without starting a delayed tx ? As it is, the MAC will be stuck busy for an unknown (to the application) amount of time, which is very confusing.

mluis1, 2018-01-17:
I forgot this point. Sorry it has been long time we have changed this behavior.

The removal of the TxDelayedTimer timer isn't yet implemented. It is on our to do list.
To remove this timer we need to refactor the OnMacCheckTimerEvent function. Currently this function requires the ScheduleTx function to return LORAMAC_STATUS_OK when the duty-cycle is restricted.

This means that if one tries to call LoRaMacMlmeRequest for the join and that the MAC has started the TxDelayTimer then the function will return LORAMAC_STATUS_BUSY.