Lora-net/LoRaMac-node #356: Many CRC_BAD error in all Received Packets by Gateway

KhaledAljehani, 2017-11-04:
I have STM32L152D + SX1276RF1JAS, and I migrated Sensor node project to that hardware. I compiled and run the project(EU868) without any modifications.

Below is a snippet of the recorded packets by the gateway:

"gateway ID","node MAC","UTC timestamp","us count","frequency","RF chain","RX chain","status","size","modulation","bandwidth","datarate","coderate","RSSI","SNR","payload"
"0000B827EBAFBA00","","2017-11-04 12:49:04.373Z", 104178947, 868300000,1, 1,"CRC_BAD", 67,"LORA",125000,"SF7"   ,""   ,-114,-12.0,"5944B0D8-9D9F1B85-7874EDBF-40F101F1-C08F002C-2BCCD3EF-28949E6A-E4158985-940F2313-C58B4A67-A9C43503-DFA643ED-C643656E-2C04DFDE-663A2694-894B52D1-DFA058"
"0000B827EBAFBA00","","2017-11-04 12:49:42.730Z", 142535435, 868100000,1, 0,"CRC_BAD",206,"LORA",125000,"SF7"   ,"4/7",-114,-11.0,"D1DB575F-C2225600-0CDA6C67-F01CF181-A39CC6F9-A0DD40DB-96FE487C-8DFFBFAE-8F6DD235-E9490B57"
"0000B827EBAFBA00","","2017-11-04 12:19:28.751Z",  95849067, 867300000,0, 4,"CRC_BAD",244,"LORA",125000,"SF7"   ,"2/3",-114,-12.0,"2F4E9087-BC8DEFFE-50E493EE-3044A074-65C95DD4-1C920B7A-1B9CE4F4-71AB7E5F-54D56195-02920772-6853AB3F-BD40D24F-28C1847A"
"0000B827EBAFBA00","","2017-11-04 12:21:04.696Z", 191792475, 867100000,0, 3,"CRC_BAD",158,"LORA",125000,"SF7"   ,"2/3",-115,-11.8,"D7E1A010-4499B8C8-7D3D"
"0000B827EBAFBA00","","2017-11-04 12:21:41.261Z", 228359323, 867100000,0, 3,"CRC_BAD",175,"LORA",125000,"SF7"   ,"1/2",-114,-11.5,"1EE8E084-A47DD510-23216D"

As you can notice above, status of all packets are CRC_Bad and coding rate is not constant. 
Later, I modified main.c, I enabled/disabled ADR and changed data rate and still the problem is the same.

Note: I brought a complete node from a company with their source code and all received packets by GW are CRC_OK without any error.

Please, help me with this issue. I tried many solutions but without any benefit.
Best regards.

reissjason, 2017-11-04:
Is the gateway using a public or private sync word?
CRC error packets with low SNR, such as these, may be caused by ambient noise.
The information cannot be trusted.

KhaledAljehani, 2017-11-05:
@reissjason 

From **global_conf.json** of the packet logger:**lorawan_public: true**
By the way, I did not make changes to the source code of the Sensor node and the gateway.

Here are additional notes:
1. As mentioned in the post, I brought a node from a company with their source code, and started TX and all packets received correctly by gateway. Note: the node has its own wire antenna.

2. With the node that causes the problem, I tried 2 different 868 antennas, and also without antenna.

3. The distance between node and gateway was 2 meters, then I increased it.

My **initial** conclusion is the problem is related to the node. 

Best regards

reissjason, 2017-11-05:
Are you able to see any transmission from the device with a spec analyzer?

reissjason, 2017-11-05:
Or current draw can be measured, should see a spike if transmission is successful.

KhaledAljehani, 2017-11-05:
Unfortunately, I do not have it.

On important point to mention is: in the code I did not modify/add the pins of the antenna because I do not know its mapping for SX1276RF1JAS module. Could it be the problem? If yes, please help with the pins.

Second point is: I tried different data rates and enabled/disabled ADR and nothing changed at all. In other words, there were no effect of the changes.

reissjason, 2017-11-05:
First things first. Get the pin mapping correct and successful tx.

Did you follow doc for SX1276 Development Kit?

KhaledAljehani, 2017-11-05:
The pin mapping is not clear in the development kit documentation. The schematic is provided but it is not clear.
Kindly, if you have clear pins declaration for the module, send it to me.

KhaledAljehani, 2017-11-05:
I found SX1276RF1JAS schematic: https://semtech.force.com/lora/LC_SearchResult?q=sx1276rf1

artgasca, 2017-11-16:
Do you found a solution? I have the same issue.

KhaledAljehani, 2017-11-16:
Unfortunately, I did not find a solution.
Anyway, this is from Semtech. It may help:

"As you can see the RSSI and SNR are VERY marginal for SF7 ; you have to investigate: 1. the truth table of the switch and if you control is properly from the Nucleo board, 2. get close to your gateway, 3. make sure the antenna is correcly connected on both sides and has the right SMA polarity 4. the correct Power Amplifier is selected in the code, with regards to your actual hardware (PA_BOOST vs. RFO)"

Let me know when you find a solution.

reissjason, 2017-11-16:
Some packets with CRC errors are expected. The LoRa demodulation can find packets out of noise. Some false positive results are expected coming from the demodulator. The CRC check is a filter for these packets, they are not necessarily an indicator of bad hardware. 

Incompatible settings between gateway and end-device or non-functioning end-device cannot be detected by a gateway.

KhaledAljehani, 2017-11-27:
Did you find a solution @artgasca ?

Gecko05, 2017-11-30:
@KhaledAljehani Do you have any sub band settings on the Gateway? If so, check out this issue https://github.com/Lora-net/LoRaMac-node/issues/202. I don't know how the node behaves when it's using EU868 but there's a way to setup the used channels with a mask.

mluis1, 2018-01-16:
Can we close this issue?

closed 2018-01-17 by KhaledAljehani

Gecko05, 2018-06-08:
@phuongNP Well the sub band has to be configured to specify in which channels you're going to work so the node doesn't try to use channels where the Gateway isn't listening. We had also a problem due to using a custom LoRa board so the firmware setting where it detects the board and selects the appropiate Tx Power didn't work properly and we had a very low transmission power, making a lot of packets fail.

phuongNP, 2018-06-15:
@Gecko05 
Thanks for your repply.
I did it. The solution is the frequency between Gateway and device configured wrongly. :)