Lora-net/LoRaMac-node #362: Min Enabled Channels in US915 and US915-Hybrid

dsmith-tek, 2017-11-14:
Upon reviewing the US region code I was wondering why the US915 and US915-Hybrid regions seem to have a different cut-off for the channel masks?

In RegionUSXXXChanMaskSet both are checked to ensure that at least 2 channels are active, but in US915-Hybrid it then calls the ValidateChannelsMask function, which limits it to 6-channels. Where does the 6 come from?

Looking through FCC 15.247 and reading OET-453039 on hybrid systems I can see no apparent reason for a 6 channel minimum. In fact OET-453039 explicitly states that "there is no minimum number of hopping channels". Any clarity on why the limit here is 6 and not 2 would be greatly appreciated.

Cheers

mluis1, 2018-02-02:
The US915-Hybrid was at beginning a temporary implementation of the US915 band to support 8 channels gateways.
When we started this project there was no server instance which implemented the channels mask handling for US915 band. 

By then we were requested to limit the minimum number of channels to 6. We will change the implementation to limit the number of channels to 2.

Nowadays all the US915 server implementations that we are aware of are able to handle the 8 channels gateways. 
Please note that in the future the US915-Hybrid implementation could be removed without notice.

dsmith-tek, 2018-02-02:
Thanks for the reply. 

Originally I had thought the Hybrid mode was just an 8-Channel variant of the full 64-Channel MAC to support let capable networks. Upon your feedback and my own review I now see that this is not the case. There seems to be some very interesting handling of channel settings and stuff that may give some weird behavior (Example, LinkAdrReq with a ChMaskCntl of 0x7# will always be invalid).

This gives us the information we need to ensure FCC compliance, and in general we intend to only use the full 64-Channel US915 Region going forward.

Cheers

closed 2018-02-02 by dsmith-tek