Lora-net/LoRaMac-node #364: Add secure element support

djaeckle, 2017-11-15:
We will add the support for a secure element for the LoRaMac stack. The update comes with API changes for the cryptographic functions.

After the update, users can decide if they want to use cryptographics in software or in hardware (with a secure element).

nestorayuso, 2017-11-30:
I am curious about different chips support:
-Trusted Objects
-Wisekey
-G+D
-Gemalto
and modules:
-Murata Type ABZ
and LoRaWAN Network Servers:
-loraserver.io
-any?

GrumpyOldPizza, 2017-11-30:
I doubt that any of those secure elements will be supportable without NDA. However it would be helpful to get an idea where the interface internally is headed towards. For one to avoid duplicate work, and then to share some thoughts ...

mluis1, 2017-12-06:
The secure element support is currently under development and should be made available once the 4.4.1 release is performed.

Our current API specification can be found on the project's wiki [Secure element API](https://github.com/Lora-net/LoRaMac-node/wiki/secure-element)
Please note that the API may change due to development constraints.

As mentioned by @GrumpyOldPizza most of the time the secure-elements support is done under NDA. Thus, our approach is to provide the building blocks to the secure-elements manufacturers to adapt their API to the one that we defined.

For users that don't have access to a real secure-element we will provide a full software emulated secure-element.

closed 2018-07-02 by mluis1