Lora-net/LoRaMac-node #366: Random number generator in boards/mcu/stm32/utilities.c is extremely non-uniform

jamesl-dm, 2017-11-22:
For instance, if you call randr(0, 7) multiple times, and arrange the output in 4 columns, you get output like this:

0, 6, 2, 0,
4, 2, 6, 4,
7, 7, 1, 0,
4, 3, 6, 4,
7, 7, 2, 0,
4, 3, 6, 4,
0, 6, 1, 0,
4, 2, 5, 5,
7, 7, 2, 1,
4, 2, 6, 5,
0, 6, 2, 0,
4, 3, 5, 4,

No matter how tall you make the columns, you'll find:
- There are no 1s, 2s, 5s, or 6s in column 1
- There are no 0s, 1s, 4s or 5s in column 2
- There are no 0s, 3s, 4s, or 7s in column 3
- There are no 2s, 3s, 6s, or 7s in column 4

In my case, this led to holes in the channel usage.
You can get a better generator at http://arxiv.org/abs/1402.6246

mluis1, 2017-11-22:
The generator it self is uniform. The `utilities.c` file implements the C 'rand' function in the same way as the GCC compiler provided random function.

Actually the uniformity is broken on the `randr` function due to the usage of the modulo operator to limit the range of the values.
But, if one lets the node run for long periods of time we will notice that in average all channels are used almost equally.

Thanks for the link.

jamesl-dm, 2017-11-23:
This is what I'm using instead. Seems to fix the issue when taking every fourth output at least.
The 64 bit generators from that paper are better, but this runs a bit faster on Cortex M0s.

```
static uint32_t miState;
int32_t rand1( void )
{
    miState ^= (miState << 13);
    miState ^= (miState >> 17);
    miState ^= (miState << 15);

    return (miState * 1332534557) & 0x7FFFFFFF;
}

void srand1( uint32_t seed )
{
    // a zero seed will not work!
    if (seed == 0)
        seed = 0x55aaff00;

    miState = seed;
}
```

closed 2017-11-23 by jamesl-dm

mrx23dot, 2022-03-04:
AES CTR can be reused to generate uniform distribution.

DevxMike, 2022-03-06:
> 

thanks a lot!