Lora-net/LoRaMac-node #367: OTAA join takes too long

Gecko05, 2017-11-22:
We're using the LoRaWAN-demo-76 and trying to join a network by OTAA to a Gateway but it takes around 8 minutes and several join requests, is this normal? Which parameters affect OTAA that I can modify to try to improve the OTAA join timing?

fahad80, 2017-11-22:
I think that's because your gateway doesn't support all channel frequencies.

Gecko05, 2017-11-28:
@fahad80 you're right, we're using sub band 4 with a Multitech gateway and it seems like the requests sent by the corresponding channel frequencies are the only ones being accepted, any idea of how could I configure sub band 4 into LoRaMac? I've something like this with the ChannelMask parameter but don't know how does it work.
'''static uint16_t multitechGatewayChannelsMask[6] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0001, 0x0000};
'''

Gecko05, 2017-11-28:
Nevermind I figured out how does the ChannelMask work, but seems like still takes several tries to join.

fahad80, 2017-11-28:
I am not sure what is sub band 4. Do you know what frequencies your gateway is accepting?

closed 2017-11-28 by Gecko05

sabas1080, 2018-02-02:
Hi @Gecko05 

How work ChannelMask?

Thanks

Gecko05, 2018-02-02:
Each byte of the channelmask controls up to 8 channels, for example to use sub band 3 you gotta have a channelmask of {0x00,0x00,0xFF,0x00,0x00...} Only the corresponding bytes of the channel sub band are enabled with the 0xFF, which means that we'd be using only 8 channels in the corresponding frequency to the sub band.