Lora-net/LoRaMac-node #369: RtcComputeTimerTimeToAlarmTick(): Is TimeoutValueError calculated correctly?

quylam925, 2017-11-29:
Hi all,

I have a little bit confused when reading the code in rtc-board.c
The function "RtcComputeTimerTimeToAlarmTick" converts the time value in ms to the value for RTC Alarm.
In this function, the error value is calculated by:
error = timeoutValue - timeoutValueTemp;       **<--- In TICK count unit**
TimeoutValueError += ( error  * 1000 );             **<--- In 1000 TICK count unit**
(**<---** indicates my comment)
then the TimeoutValueError value is compared with (RTC_ALARM_TICK_DURATION * 1000)  **<--- In us unit**
How can we compare like that???
I think the error value should be changed to us unit instead of Tick count unit before comparing with (RTC_ALARM_TICK_DURATION * 1000)
Therefore, the code should be changed from:
**error = timeoutValue - timeoutValueTemp;** 
to
**error = (timeoutValue - timeoutValueTemp) * RTC_ALARM_TICK_DURATION;** 
Is that correct?
Could somebody explain this to me please. It would be really appreciated,
Thanks

mluis1, 2018-02-05:
Can we close this issue?

We think that the implementation is correct and validated.

quylam925, 2018-02-06:
I have closed the issue.
Thanks

closed 2018-02-06 by quylam925