Lora-net/LoRaMac-node #371: Class C Rx1 window timeout issue!

fahad80, 2017-11-30:
It's a trivial issue. During Class C, we set the Rx1 window timer but not Rx2 Window timer. When we receive the Rx timeout interrupt from semtech IC, OnRadioRxTimeout() is called, and of course, this interrupt is for Rx1 Window.
,,,
static void OnRadioRxTimeout( void )
{
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }

    if( RxSlot == 1 )
    {
        if( NodeAckRequested == true )
        {
            McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;
        }
        MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;

        if( LoRaMacDeviceClass != CLASS_C )
        {
            LoRaMacFlags.Bits.MacDone = 1;
        }
    }
    else if( RxSlot == 0 )
    {
    	if( NodeAckRequested == true )
		{
			McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX1_TIMEOUT;
		}
		MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX1_TIMEOUT;

		if( TimerGetElapsedTime( AggregatedLastTxDoneTime ) >= RxWindow2Delay )
		{
			TimerStop( &RxWindowTimer2 );
			LoRaMacFlags.Bits.MacDone = 1;
		}
    }
}
,,,

Now the problem is as we are class C hence OnRxWindow2TimerEvent() is called and that function changes RxSlot  value to 1 (but the interrupt was for RxSlot 0). So, now we get wrong Mcps and Mlme Confirm.Status, LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT instead of LORAMAC_EVENT_INFO_STATUS_RX1_TIMEOUT. I think moving the class checking after slot checking will fix the issue!

djaeckle, 2017-12-01:
Hi fahad80,

thanks for your report. We will double check it.

bwesen, 2018-01-04:
I'm currently bugged by an issue where every second Class C packet sent from the GW to the node is lost silently, it seems it mostly happens for large packets (DR10, in US915, which can be 240 bytes and spend about 490 ms on air). I can see in the gateway log that (for example) 4 packets are sent, but on the node, the second packet is never detected. The gateway sends them as fast as it can, as far as I can see. Sending smaller packets, like 50 bytes (on the same DR!) seems to work fine.

So I was searching here for any other bugreports related to tightly scheduled class C received packets, could this be a related issue you think?

mluis1, 2018-02-08:
@bwesen the issue that you are reporting is not related to this issue. We the close the issue.

closed 2018-02-08 by mluis1