Lora-net/LoRaMac-node #373: LoRA SX1276 can't send more than 64 byte

artuguy, 2017-12-01:
Hi, Everyone:
            
      I've ported sx1276 driver to the stm32f103c8,  havn't use loramac yet.  2 node work in LoRa mode,  one in continue  recieve mode,  one in send mode,  when I send < 64 bytes, everyting is OK, but when i send more than 64, ie 65,  it can't work,  the txdone  is trigger,  but  reciecve node can't indictate packet rxdone.

How dose it do?

Thanks

fahad80, 2017-12-01:
Because:
```
#define LORAWAN_APP_DATA_BUFF_SIZE                 64
```

artuguy, 2017-12-02:
fahad80：

           thank you， which file does the
                                 #define LORAWAN_APP_DATA_BUFF_SIZE                 64
reside？

fahad80, 2017-12-04:
"lora.c"

mluis1, 2018-01-16:
Can we close this issue?

artuguy, 2018-02-04:
OK,  close it.

closed 2018-02-05 by mluis1