Lora-net/LoRaMac-node #376: Class C and JOIN issue

fahad80, 2017-12-06:
If I set my device as Class C before initiate join procedure, after a successful JOIN the radio is in sleep mode. 
In other words, I will have to set my device to Class C again. 

I've tracked down the issue. Once I receive a rxDone interrupt LoRaMac execute onRadioRxDone(). That function puts the radio in sleep. At the end the function starts the OnMacStateCheckTimer. 
OnMacStateCheckTimerEvent() checks for class C only if LoRaMacFlags.Bits.McpsInd is set.
```
if( LoRaMacFlags.Bits.McpsInd == 1 )
    {
        if( LoRaMacDeviceClass == CLASS_C )
        {
        	// Activate RX2 window for Class C
            OnRxWindow2TimerEvent( );
        }
        if( LoRaMacFlags.Bits.McpsIndSkip == 0 )
        {
            LoRaMacPrimitives->MacMcpsIndication( &McpsIndication );
        }
        LoRaMacFlags.Bits.McpsIndSkip = 0;
        LoRaMacFlags.Bits.McpsInd = 0;
    }
```
As this was a JOIN request the flag isn't set hence RX2 indow is never open.

So my question is, is this a bug or m I supposed to change class only after a JOIN?

svenKautlenbach, 2017-12-06:
My CLASS C is working correct. Although what I discovered is that the radio is put to DR_3(SF9) instead of DR_0 for RX2 winow. This is EU868 band. I manually set the window dr parameter to DR_0.
[image]

I have not had time to track down why (maybe an error from human side - me) but maybe this helps.

fahad80, 2017-12-06:
I never said Class C isn't working. It's working. But I had to change to class C after I JOINED a network. If I changed my class using LoRaMacMibSetRequestConfirm() before I join a network, it doesn't work. What order did you follow?

fahad80, 2017-12-06:
Maybe I wasn't clear enough. My Class C works. I am asking a different question. Simply put:

1) Change Class to C ------ JOIN------> Doesn't work
2)Class A--JOIN--Change Class to C--> Works

Successful join puts Radio into sleep, even if you are class C. I beilieve its a bug.

mluis1, 2017-12-06:
@fahad80 If you are using OTAA activation you must set the device in ClassC after the join procedure succeeds.
The device is supposed to always start as a ClassA device. Thus the device class parameter gets reset when performing the Join procedure.

When using ABP activation the issue isn't seen.

We will update the application examples as soon as possible.

fahad80, 2017-12-06:
@mluis1 Thanks a lot. That's what I needed to know. Now I can go to sleep peacefully.

closed 2017-12-06 by fahad80