Lora-net/LoRaMac-node #377: AckTimeoutTimer during Class C

fahad80, 2017-12-06:
Several issues:
1) In OnRadioTxDone() why are we starting AckTimeoutTimer in Class C even if NodeAckRequested is false? If I send an unconfirmed packet, what would be the purpose of AckTimeoutTimer?
```
static void OnRadioTxDone( void )
{
    . . . . . . . .
    . . . . . . . .
    if( ( LoRaMacDeviceClass == CLASS_C ) || ( NodeAckRequested == true ) )
    {
        getPhy.Attribute = PHY_ACK_TIMEOUT;
        phyParam = RegionGetPhyParam( LoRaMacRegion, &getPhy );
        TimerSetValue( &AckTimeoutTimer, RxWindow2Delay + phyParam.Value );
        TimerStart( &AckTimeoutTimer );
    }
```

2)With that, @djaeckle 's commit https://github.com/Lora-net/LoRaMac-node/commit/32c9b19ad5059256dd92a85dec373a0f9b23b995 broke Class C (receiving data on RX2 window). Here is how:
After a successful JOIN, I can receive data from NS. All good. But as soon as I send a packet from node, it stops receiving packets from NS on RX2 window. I traced down the issue at OnRadioRxDone():

```
 // Verify if we need to disable the AckTimeoutTimer
    CheckToDisableAckTimeout( NodeAckRequested, LoRaMacDeviceClass, McpsConfirm.AckReceived, AckTimeoutRetriesCounter, AckTimeoutRetries );

    if( AckTimeoutTimer.IsRunning == false )
    {
    	// Procedure is completed when the AckTimeoutTimer is not running anymore
        LoRaMacFlags.Bits.MacDone = 1;

        // Trig OnMacCheckTimerEvent call as soon as possible
        TimerSetValue( &MacStateCheckTimer, 1 );
        TimerStart( &MacStateCheckTimer );
    }
```

Although CheckToDisableAckTimeout() turned off AckTimeoutTimer but AckTimeoutTimer.IsRunning is still true. There is only one way AckTimeoutTimer.IsRunning is still true after a TimerStop(). From the TimerStop() implementation it seems that flag is set to false only if it is the TimerListHead. Otherwise the obj is just removed from the list. So although the timer is "no longer in the list", that flag is still true!!!!

AckTimeoutTimer.IsRunning was set to true during the packet send from node to NS. Although AckTimeoutTimer was stopped after the predefined time that flag is still true ( you can verify it in OnAckTimeoutTimerEvent(). AT the end of that function the flag was still true ).

Am I making any sense?

djaeckle, 2017-12-08:
Hi fahad80,

thanks for your report! I will double check your finding. At least my simulations provide a pass for my commit - however, the timer implementation is abstracted here. I will come back to you soon.

mluis1, 2018-02-02:
Could you please try the following patch?
Please let us know if it solves the issue.

```patch
diff --git a/src/system/timer.c b/src/system/timer.c
index 526e0e3..1aa6063 100644
--- a/src/system/timer.c
+++ b/src/system/timer.c
@@ -280,9 +280,9 @@ void TimerStop( TimerEvent_t *obj )
 
             remainingTime = obj->Timestamp - elapsedTime;
 
+            TimerListHead->IsRunning = false;
             if( TimerListHead->Next != NULL )
             {
-                TimerListHead->IsRunning = false;
                 TimerListHead = TimerListHead->Next;
                 TimerListHead->Timestamp += remainingTime;
                 TimerListHead->IsRunning = true;
```

mluis1, 2018-02-08:
The proposed fix should solve the issue. As no feedback is provided we close the issue.

closed 2018-02-08 by mluis1