Lora-net/LoRaMac-node #379: How the sleep mode works in STM32L0?

AlstomJT, 2017-12-07:
Hi,

I’m really new in the use of the Lora end-devices and I have some troubles to achieve a low consumption in sleep mode (I get 0,13mA).

My lora end device has a CMWX1ZZABZ-078 micro-controller with a Chipset: Semtech (SX1276) + STM (STM32L) ) and I have also the B-L072Z-LRWAN1 LoRa Discovery kit (to understand better how things work). I use the example “LRWAN-NS1” from the I_CUBE_LRWAN which is an example to start a simple use of a LORA en device.

It’s been a couple of weeks that I’m trying to understand the sleep mode to be sure that I’m able to go lower than 0,13mA for a longer time life of my batteries. I appreciate any help you can give me to keep going with this end device.

Best Regards,
Jorge.

nestorayuso, 2017-12-07:
I think you are using the source code from http://www.st.com/en/embedded-software/i-cube-lrwan.html
and not the source code from https://github.com/Lora-net/LoRaMac-node

There is one difference in CMWX1ZZABZ-078 compared to other modules; there is a TCXO instead of a Crystal, you must power it on and off by software, make sure the TCXO is powered off before the MCU goes to low-power.

AlstomJT, 2017-12-08:
You are right, the source code that I'm using is from the ST site. As I really new in this field, I thought that the example given by ST was adapted to my need (configure the CMWX1ZZABZ-078 to communicate with a low power consumption). So, I did some modification in that source code to communicate with the LORA server. I succeeded the communication but I'm not in low power consumption. I took a quick look in the source code from https://github.com/Lora-net/LoRaMac-node and I didn't found and example close to my need.

Thanks for that advice, I'm going to do some research about it and try to fix/modify the source code. I really appreciate your help. I'm really open to listen any advice related to this topic.

mluis1, 2018-01-16:
Can we close this issue?

closed 2018-01-17 by AlstomJT