Lora-net/LoRaMac-node #383: Problem with UART after RtcEnterLowPowerStopMode

dayar, 2017-12-14:
Hi, I have problems with the UART after call RtcEnterLowPowerStopMode and wakeup.
I set a periodic timer and enter in low power mode.
When timer fires interrupt, I call a function to send some serial data, but serial data is junk.
The function initializes uart and fifo buffers.

Any idea what could be happening?
Thanks in advance

fahad80, 2017-12-14:
Check the clock. STOP mode changes the clock. If the clock you are thinking that's running and the actual clock isn't the same, you'll have baud rate mismatch. Hence, junk data.

dayar, 2017-12-15:
Hi, I checked the clock, and seems thats is ok.
 I tested the problem more, I've seen that if I set the baudrate to 115200b, after wape up, the serial data transmited is correct. But if I set the baudrate to lower values the serial data is junk.
In all lower baudrates , if I sen data before enter in low power stop mode, the data transmited is correct, the problem is caused after wake up from stop mode.

I appreciate the help to solve the problem.

fahad80, 2017-12-19:
How are you checking the transmitted data? Using some terminal software or using logic analyzer?

mluis1, 2018-01-16:
Can we close this issue?

closed 2018-02-05 by mluis1