Lora-net/LoRaMac-node #385: RegionXXXAlternateDr() enables purposely disabled channels.

catsup, 2017-12-18:
Hello,

Using the US915 MAC, v4.4.0 branch (but the code is identical on devel branch).

my application sets up a channel mask to ensure that the MAC only uses channels that are actually supported by the gateway, namely channels 0-7:

	MibRequestConfirm_t req;
	LoRaMacStatus_t rc;
	uint16_t mask[6] = { 0 };

	mask[0] = 0x00ff;		// enable only channels 0..7

	req.Type = MIB_CHANNELS_MASK;
	req.Param.ChannelsMask = mask;

	rc = LoRaMacMibSetRequestConfirm( &req );

This returns a success status, so I assume the mib request was executed successfully.

However, when trying to perform an OTAA join, the MAC sends join requests over channels that are not currently enabled.

This appears to be because of the following regional code, which forces channels 64-71 to be enabled:

int8_t RegionUS915AlternateDr( int8_t currentDr )
{
    static int8_t trialsCount = 0;

    // Re-enable 500 kHz default channels
    ChannelsMask[4] = 0x00FF;

    if( ( trialsCount & 0x01 ) == 0x01 )
    {
        currentDr = DR_4;
    }
    else
    {
        currentDr = DR_0;
    }
    trialsCount++;
    return currentDr;
}

The same issue appears in some of the other regions.

djaeckle, 2017-12-20:
Hi catsup,

thanks for your report. I think this is related to issue #281. We will discuss it and come back to you.

mluis1, 2018-01-16:
We close this issue. Please refer to the latest answer provided on issue #281.

closed 2018-01-16 by mluis1

catsup, 2018-01-16:
Hello mluis,

Please re-open this issue.

#281 was closed on the basis that the ADR mechanism may re-enable all default channels, however this issue here is specifically about the forced re-enabling of DR4 channels for the join request alternate datarate mechanism (which was also mentioned in #281).

This forced re-enabling of DR4 channels is an issue as the gateway (Kerlink) only supports DR0 channels 0-7, and will not receive join requests on DR4 channels.

kjendal, 2018-01-16:
Regardless of how any particular gateway works, re-enabling of DR4 is required by specification.  Also, the Kerlink gateway does support DR4.  The impact of not re-enabling all channels is fatal to the device (as local channel plans may support any portion of the 72 channels available).  Enabling channels that are not available on the local channel plan is not fatal, merely a SMALL inconvenience in increasing the number of JoinRequests which must be sent before success.  As Join occurs very infrequently, this is not an operational issue.
-Dave Kjendal, Senet

catsup, 2018-01-16:
Hi Dave,

thanks for your comment.

I checked the 1.0.2 regional specification but did not see where the 1.0.2 regional specifications it mandates to re-enable DR4 channels on a join request -  could you point me to the corresponding paragraph ?

When I set a channel mask (or any other parameter for that matter) through a mib request, I don't really expect that parameter to be changed without notice.

I'm not sure my gw supports dr4 (it's one of the old 8-channel kerlink boxes, not the larger 64-channel one), but will doublecheck to make sure it's not simply a matter of configuration on the network side (or me being impatient in case it supports only one DR4 channel which I never saw used).

Regards,
Marc