Lora-net/LoRaMac-node #386: Enable to make the Ping-Pong in FSK works - stuck in FSRx

jeanguerin, 2017-12-21:
Hello,

We have adapted your source code to work on the EFM32LG of Silabs with SX1272MB2xAS. The Ping-Pong example works fine in LoRa modem, but not at all in FSK modem.
When I look at the opMode, it never goes to Rx (opmode:5), it seems to be stuck in FSRx (opmode:4).
I tried to put one board in Tx and one board in Rx, with the Tx launching a Ping every second. It takes almost 30 s to the Rx to switch from FSRx to Rx. The preamble is then recognized. And then, sometimes, it recognized the syncAdress.
The settings are those included in the ping-pong. 

Could you help me please?

nestorayuso, 2017-12-21:
https://github.com/imec-idlab/LoRaMac-node

this is a EFM32 port of the LoRaMac-node project, maybe can be helpful for you

mluis1, 2018-02-05:
Can we close this issue?

closed 2018-02-08 by mluis1