Lora-net/LoRaMac-node #388: AckTimeout event During Class C

fahad80, 2017-12-22:
In OnRadioTxDone() why are we starting AckTimeoutTimer in Class C even if NodeAckRequested is false? If I send an unconfirmed packet, what would be the purpose of AckTimeoutTimer?
```
static void OnRadioTxDone( void )
{
    . . . . . . . .
    . . . . . . . .
    if( ( LoRaMacDeviceClass == CLASS_C ) || ( NodeAckRequested == true ) )
    {
        getPhy.Attribute = PHY_ACK_TIMEOUT;
        phyParam = RegionGetPhyParam( LoRaMacRegion, &getPhy );
        TimerSetValue( &AckTimeoutTimer, RxWindow2Delay + phyParam.Value );
        TimerStart( &AckTimeoutTimer );
    }
```

twowinter, 2018-01-03:
same question, need help

djaeckle, 2018-01-11:
Hi fahad80,

thanks for your question. We start the timer as we need an event to stop the uplink procedure. For class A, we have timeouts for both of the reception windows - so we can detect when a TX / RX procedure has been finished. 

However, in class C, the second window is continuous configured without a reception timeout. Thus, we use the AckTimeoutTimer to notify the MAC that the procedure has been finished.

fahad80, 2018-01-11:
Hi djaeckle ,

Correct me if I'm wrong, `OnAckTimeoutTimerEvent()` just set `LoRaMacFlags.Bits.MacDone` during Class C. Can we not do that right after 1st Rx Window Timeout (inside `OnRadioRxTimeout()`)? It is not critical I know, I am just curious. Thanks.

djaeckle, 2018-02-08:
Hi fahad80,

thanks for your feedback. I think an implementation according your proposition would also be possible. However, we would need to update the handling of the MAC layer regarding the ACK timeout. Currently, the MAC considers a procedure as finished, when the ACK timeout fires. Especially, this is related to uplink repetitions - according to the specification, the node has to wait for the ACK timeout before it is allowed to send another frame.

Please let me know if your question has been answered and if we can close this issue.

fahad80, 2018-02-08:
@djaeckle Thanks for the answer. I think I understood it. Closing the issue.

closed 2018-02-08 by fahad80